FrSkyRC/RF_Feedback_Group #26: RB25s does not switch into automatic mode in case of failsafe , channel 14 = -100%

Kreisflieger, 2024-08-26:
I am using a RB25S in Quick-Mode with an Archer Plus R8 connected via FBus. I use a X18SE with Ethos 1.5.3.
Firmware of RB25S, Archer Plus R8 and RB25-LUAS are all on latest versions.
To setup stabilization I set channel 13 (gain) on a pot and channel 14 (stab-mode) on a switch.
Everything works fine in Stabilization and Automatic Mode, as expected. 

Now I want to activate automatic mode when the receiver goes into failsafe.
Therefore I set channel 14 in the LUA config page 3/3 (failsafe settings) to -100%. Gain is set to 70% ( appr. 1750 micro sec. ) This should activate automatic mode with a good gain setting.
Unfortunately, the RB25S does not switch to automatic mode when receiver is in failsafe.

Checking the servo output of the RB25S on channel 14, it is switched to -100% which equals automatic mode.
But the RB25S is still in stabilization “off mode”.
I also set the receiver channel 14 to -100%. No success, either.
So I assume that there is a missing logical connection somewhere in the setup: Ethos, config LUA or RB25 firmware.
As known, all current S-receivers switch into automatic mode, if failsafe on channel 14 is set to 
-100%. 
Why not the RB25S ??

Kind regards

Kreisflieger, 2024-08-26:
Hope, the RF team will now take care about this issue.

SmartTommy, 2025-01-04:
[RBX5.zip](https://github.com/user-attachments/files/18305380/RBX5.zip)

Please conduct a comprehensive test to see if it can meet your needs. 

use the ethos 1.6.0 and TX & RX 3.0.1 and the new LUA(included in the packet)

Kreisflieger, 2025-01-04:
@ SmartTommy
Thanks for providing the files.
I tested RBX5.zip with:
X18RS, Ethos 1.6.0, TD-ISRM RS 3.0.1
Archer Plus R8, 3.01, connected via FBus
RB25S with your provided Firmware: RB25S_3.0.1RC2.frsk and LUA RB25_S ( RBx5)
See Info:
 
 
[image]
[image]

Using the device config:
 
[image]

[image]

Selecting RB25S calibration does not show the drawing of the RB25S as provided by the
 cali folder:
 
[image]

But the calibration does work
 

Test result of the RB25S is  as expected
As the failsafe for channel 14 was still set ( on  page 1-3 of RB25(S) ) to 890 micro sec, switching off the transmitter, the RB25S switched instantly into automatic mode, perfect ! That what I was looking for.

But, I was not able to alter the failsafe setting of other channels. As soon as I click on the field: **Record current value**, I get a LUA warning **red triangle**  and the value of the channel resets to the before value.
 
[image]

Clicking on the triangle shows:
 
[image]

To sum up my findings:
-	The general function of the RB25S for flying is now as expected.
LUAS have to be improved:
-	In calibration mode, the drawing of the RB25S should be shown
-	Setting the failsafe values should not create a red triangle warning and the new failsafe value should be stored

Hope my findings are helpful for you to improve the LUAs.

Many regards 
Stephan

Kreisflieger, 2025-01-04:
While I was still testing the new version I got an extended LUA Script warning:

[image]

Nicholas-Luoyi, 2025-01-06:
Fixed the first error. But I cannot reproduce the second error

Kreisflieger, 2025-01-06:
Thanks Nicolas,

if you can provide the updated LUAs I am happy to further test.

Regards

Nicholas-Luoyi, 2025-01-06:
Will upload 2.1.2 later. Please update it in Suite

Nicholas-Luoyi, 2025-01-07:
> go to page 3-3 of config, directly type in a value in the left hand microsecond field, e.g. 1000

I guess you mean right hand?

Kreisflieger, 2025-01-07:
yes, sorry for the mix up.

[image]

and the corresponding red triangle message:
[image]

Kind regards Stephan

Nicholas-Luoyi, 2025-01-07:
Still cannot reproduce. But I can propose a fix. Please try.

[RB25_S.zip](https://github.com/user-attachments/files/18329380/RB25_S.zip)

Kreisflieger, 2025-01-07:
tested your latest version:

if I change the channel 1 value field ( e.g. to 1000 ) with the stick in middle position and pressing the "record current value" the value field jumps to 1500, no red triangel.

This is not too helpful but it avoids the red triangle.

But instead may I suggest to change the value field into a **read only field** and display the current value of that channel to be able to set the channel to the desired value, when pressing "Record current value"

Kind regards  Stephan

Kreisflieger, 2025-01-07:
Dear Nicholas,
to further dig into this behavour, can you please send the RB25 firmware, which you are using for your test/ LUA development.

Then we will be sure that I have the the correct version.

Thanks in advance   Stephan

Nicholas-Luoyi, 2025-01-08:
They are independent from each other. I am responsible for Lua scripts. And @SmartTommy is responsible for the device firmware

Kreisflieger, 2025-01-08:
Thanks, good to know.

I will continue to test, as I do not understand why my X18RS is behaving differently to your setup.

SmartTommy, 2025-01-08:
> Thanks, good to know.
> 
> I will continue to test, as I do not understand why my X18RS is behaving differently to your setup.

[RB35_3.0.1_RC1.zip](https://github.com/user-attachments/files/18344189/RB35_3.0.1_RC1.zip)

closed 2025-01-09 by SmartTommy