FrSkyRC/RF_Feedback_Group #4: X20R with TD-ISRM 1.1.14 three times same telemetry value and error messages

glipski50, 2024-08-26:
Using
X20R with TD-ISRM FW 1.1.4 Ethos 1.5.13
TW R12 with FW 1.0.9
Neuron II 60A with FW 11_5

TW R12 Pin 12 is set to S.Port
Yellow wire of the Neuron II (S.Port/FBUS) is connected to Pin 12

In telemetry I get ESC Voltage and ESC Current shown three times as three pairs with three different addresses. All other telemetry values are shown as normal only one time.
 
[image]

the upper pair has address 0D 0B50
the middel pair has address 0A 0B50
the lower pair has adress 17 0B50

the upper pair Voltage / Current and the lower pair Voltage Current frequently get lost and come back. The middle pair is shown constantly like in the examples below. When they disappear or come back it is always as a pair Voltage / Current

[image]

[image]

That goes along with constant messages Sensor Lost when one pair gets lost or Sensor Conflict when one pair comes back

SmartTommy, 2024-08-26:
Hi @glipski50 , i repeated the phenomenon you described if only connect the S.PORT line, could you please try to also connect the PWM/SBUS line to your receiver when you test ?

glipski50, 2024-08-26:
Hi SmartTommy, I connected the PWM line to the throttle pwm out put, deleted all sensors and discovered sensors agin. Now ESC voltage and ESC current are shown correctly only once. But I got a sensor conflict message with 5 minutes. I only have the Neuron II 60 connected to the TW R12 receiver, no other sensors.

SmartTommy, 2024-08-27:
We attempted to reproduce the phenomenon you described, but everything worked properly when connected  the throttle pwm to the ESC.  do you have anther ESC or can you start the motor to run?

glipski50, 2024-08-27:
@SmartTommy, I do not have another ESC to test. Also I cannot start the motor since ESC / Motor are not mounted in my model. So far I wanted to use the ESC Neuron II 60A only to power up the receiver TW R12 and to see telemetry data for voltage and current, secifically BEC of Neuron II, while I am building up a powered glider with 10 servos in wing, tail.

Could you reproduce to get the telemetry data ESC voltage and ESC current three times when the PWM line of the ESC is not connected? As I wrote above in my initial posting why is the ESC voltage and ESC current show three times with three different PhysIDs
one time 0D 0B50 unstable display with sensor lost and sensor conflict messages
one time 0A 0B50 stable
one time 17 0B50 unstable display with sensor lost and sensor conflict messages

I do not know where the Phys ID sensor adresses are generated, ESC, TW R12 or TD-ISRM-Module. Why would the PWM input of the ESC, if connected or not connected, have impact on the telemetry data?

This strange behavior did not happen with the initial TD-ISRM version of my X20R. It only came up after I have updated the TD-ISRM to the new version 1.1.14. Unfortunately I cannot go back to the previous TD-ISRM version since it is not available to download on the Frsky-rc webpage.

I believe that there is an issue with the latest TD-ISRM firmware update. 
This issue https://github.com/FrSkyRC/RF_Feedback_Group/issues/36 alo might be related to this.

SmartTommy, 2024-08-27:
[FW-TD-ISRMPro_v1.1.13.zip](https://github.com/user-attachments/files/16758262/FW-TD-ISRMPro_v1.1.13.zip)
 here:https://www.frsky-rc.com/tandem-x20-pro-aw/

SmartTommy, 2024-08-27:
> @SmartTommy, I do not have another ESC to test. Also I cannot start the motor since ESC / Motor are not mounted in my model yet. I am still in the process of building up the model. So far I used the ESC Neuron II 60A only to power up the receiver TW R12 and to see telemetry data for voltage and current, specifically BEC of Neuron II, while I am installing the 10 servos wing and tail and testing power current / consumtion load of the servos.
> 
> Could you reproduce to get the telemetry data ESC voltage and ESC current three times when the PWM line of the ESC is not connected? As I wrote above in my initial posting why is the ESC voltage and ESC current show three times with three different PhysIDs one time 0D 0B50 unstable display with sensor lost and sensor conflict messages one time 0A 0B50 stable one time 17 0B50 unstable display with sensor lost and sensor conflict messages
> 
> I do not know where the Phys ID sensor adresses are generated, ESC, TW R12 or TD-ISRM-Module. Why would the PWM input of the ESC, if connected or not connected, have impact on the telemetry data?
> 
> This strange behavior did not happen with the initial TD-ISRM version of my X20R. It only came up after I have updated the TD-ISRM to the new version 1.1.14. Unfortunately I cannot go back to the previous TD-ISRM version since it is not available to download on the Frsky-rc webpage.
> 
> I believe that there is an issue with the latest TD-ISRM firmware update. This issue #36 also might be related to the issue I reported.

The ESC  switch between S.PORT/FBUS until find the throttle PWM, so it will report the telemetry randomly.

glipski50, 2024-08-27:
Thanks for the quick response. I will continue later since I am not at home for the next couple of days.

SmartTommy, 2024-08-29:
[TD-ISRMPro_1.1.15_20240828_Beta.zip](https://github.com/user-attachments/files/16775525/TD-ISRMPro_1.1.15_20240828_Beta.zip)

please test this Beta firmware.

glipski50, 2024-08-29:
I will test the betafirmware this weekend

glipski50, 2024-08-31:
I did some tests on the bench with the TD-ISRMPro_1.1.15 beta firmware. There is no more a sensor conflict message at power up of the receiver and no sensor lost message. Telemetry works fine which anyway was a probllem of not having connected the pwm input of the esc.
 
I assume there will be a ISRM firmwara update in the near future. Should i use the beta firmware for flying or should I wait for the official update?

Thanks for your support

SmartTommy, 2024-08-31:
> I did some tests on the bench with the TD-ISRMPro_1.1.15 beta firmware. There is no more a sensor conflict message at power up of the receiver and no sensor lost message. Telemetry works fine which [...]

The release version software need about two week to test.