FrSkyRC/RF_Feedback_Group #6: X20R and TW SR8 OTA update error

glipski50, 2024-08-26:
Looks like OTA update of the TW SR8 to the latest firmware 1.0.9 does not work.

I got a TW SR8 receiver today which I wanted to update via internal OTA  to the actual FW 1.0.9. Transmitter is a X20R with Ethos 1.5.12.

I copied the firmware file tw_sr8_1.0.9.frsk to the transmitter and then started update via internal OTA

[image]

The target receiver is shown to for selection

[image]

After selecting the target receiver I get the error message "OTA update error - receiver  not supported"

[image]

I could update the TW SR8 via the S.Port with the X20R port successfully and without any problem and I also could calibrate the TW SR8 with the actual Stab Config Lua 3.0.0 .

I then tried to flash the TW SR8 again via OTA and still got the same error a desrcibed above.

ilioSS53, 2024-08-26:
Good day all,
As I understand it well Bertrand labbeled as a bug.
So it has his attention and we can expect a fix for this?
Is this correct?
Kind regards,
Theo

rburrow87, 2024-09-02:
Ah, I didn't think to look in here before submitting mine.

I think it's clear this is more of an Ethos bug since it breaks after 1.5.11, unless something was added to both RF and Ethos at the same time.

This really needs to be fixed. It's crippling if you take an aircraft to the field and find you haven't had it out since one of the last major RF updates that requires updating everything.

ronron84, 2024-09-02:
hi, 
i use a td r10 with a x20 pro eu lbt, i can bind it but i don't get a good signal, it is erratic, leds on receiver are  erratic too,i try to update by ota and smart port the receiver is not recognized by the radio, impossible to update.

SmartTommy, 2024-09-03:
Due to the addition of identity verification function in the new software, each receiver's software needs to be separately authenticated, otherwise it cannot  OTA.

Please test the files in the attachment, we will complete the testing and release them as soon as possible.
Other models of receivers will also be available tomorrow
[TD_Receiver_Beta.zip](https://github.com/user-attachments/files/16846171/TD_Receiver_Beta.zip)

ronron84, 2024-09-03:
Hi,
i've tested  files with TD 12 the file  with X20R👍 
1 i copy the file in my firmware  sub directory.
2 i launch the file 
3 i select OTA with internal module 
4 it ask me to bind the receiver
5 i turn off and on the receiver, bind ok
6 the update start and finished perfectly.
the signal is straight and perfect, smartport works perfect too !
here the vidéo : https://youtu.be/tIq4Fuez_To
PERFECT JOB

glipski50, 2024-09-04:
> Due to the addition of identity verification function in the new software, each receiver's software needs to be separately authenticated, otherwise it cannot OTA.
>
> Please test the files in the [...]

Will there be new firmware also for TW receivers i.e. TW SR12?

SmartTommy, 2024-09-06:
> @SmartTommy
> 
> > Due to the addition of identity verification function in the new software, each receiver's software needs to be separately authenticated, otherwise it cannot OTA.
> > Please test the files in the attachment, we will complete the testing and release them as soon as possible. Other models of receivers will also be available tomorrow [TD_Receiver_Beta.zip](https://github.com/user-attachments/files/16846171/TD_Receiver_Beta.zip)
> 
> Will there be new firmware also for TW receivers i.e. TW SR12?

Yes

bsongis-frsky, 2024-09-06:
Duplicate deleted https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/4473

ronron84, 2024-09-07:
I have another question : when the signal was erratic due to problem below, i didn't have any message, just a mouvement of the signal 0 to 100% on the radio, i've only got a message when i plug an OXS sensor (gps, mp6050). if the radio needs to detect a multi  data sensor error to send a message, (we also try with a frsky vario and no message), it can be dangerous. i think. What are the condition to get a warning message?

closed 2024-09-13 by SmartTommy

i3dm, 2024-10-11:
> Collaborator

why is this addition needed?
everythin worked fine without it so far.
do you think pilots should reflash all their recievers again? this is a huge burdon for the pilots!

i3dm, 2024-10-11:
> Due to the addition of identity verification function in the new software, each receiver's software needs to be separately authenticated, otherwise it cannot OTA.
>
> Please test the files in the [...]

why is this addition needed?
everythin worked fine without it so far.
do you think pilots should reflash all their recievers again? this is a huge burdon for the pilots!

SmartTommy, 2024-10-12:
> > Due to the addition of identity verification function in the new software, each receiver's software needs to be separately authenticated, otherwise it cannot OTA.
> > Please test the files in the attachment, we will complete the testing and release them as soon as possible. Other models of receivers will also be available tomorrow [TD_Receiver_Beta.zip](https://github.com/user-attachments/files/16846171/TD_Receiver_Beta.zip)
> 
> why is this addition needed? everythin worked fine without it so far. do you think pilots should reflash all their recievers again? this is a huge burdon for the pilots! i personally have over 30 Rxs mounted and flashing all of them would take days.

Because a small number of users upgrade to the wrong file，adding this verification feature will be able to avoid this situation.
Actually, if the current firmware works well for you, don't have to update it every time

ilioSS53, 2024-10-12:
Please be more precice TW or TD

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

i3dm, 2024-10-12:
> Because a small number of users upgrade to the wrong file，adding this verification feature will be able to avoid this situation. Actually, if the current firmware works well for you, don't have to [...]

That's incorrect.
My system was perfectly fine and today I opened a new TD R10 and wanted to update it to latest fw and it fails on "unsupported Rx".

So now I have to start manual updates of every new TDR10??