FrSkyRC/RF_Feedback_Group #7: New 3.0 LUA stab tool bug - bricks the RB35s

noamkrief, 2024-08-26:
If you max out the data rate in "config", I think 25.5 was maximum.
It bricks the RB35S.
Can no longer access any of the LUA pages. All grayed out.

Please provide a description of what "data rate" does, as I did not see a description in the PDF manual.
Thanks :)

Re-flashing the RB35S does not rescue it.
It will be missing the Batt1 Batt2 telemetry items and none of the LUA scripts will work.'

SmartTommy, 2024-08-27:
please try the default frk, it will restore the device to factory settings. after upgrade the firmware, repower it, then it will works.

and then flash the last firmware  https://www.frsky-rc.com/rb-35-rb-35s/

it will works normal after this.

NOTE: Date Rate is the Response frequency，255 means 25.5S, it is too long. please keep it within 10(1S), recommend using 3

noamkrief, 2024-08-27:
@SmartTommy 
Can you explain again?
Response of what? Response of telemetry items?

In the LUA we have 25.5 as maximum. Is that 2.5 seconds???
You recommend 3. Meaning 0.3 seconds?
Can you explain why you recommend using (3)?
Thank you!

SmartTommy, 2024-08-28:
> @SmartTommy Can you explain again? Response of what? Response of telemetry items?
> 
> In the LUA we have 25.5 as maximum. Is that 2.5 seconds??? You recommend 3. Meaning 0.3 seconds? Can you explain why you recommend using (3)? Thank you!

255 means 25.5 seconds, it means RB35S send the telemetry data every 25.5S, too slowly.   you can set the value from 1~10, The shorter the time, the more frequent the RB35S replies, and the higher the real-time data performance, but it will occupy more bandwidth.For some receivers with low transmission rates(such as ACCST receiver), a too fast reply speed can actually result in data not being transmitted in time , so 3 is appropriate, neither too fast nor too slow. if you use TW/TD Receivers you can used 1(0.1S),because it has a sufficiently high transmission rate.

next LUA version will limit this value from 1 to 10.

noamkrief, 2024-08-28:
@SmartTommy 
Thanks for the clarification.
From my other ticket attempts, you are aware that the RB35S already by itself occupies most of the telemetry bandwidth right?
And adding any additional sensors such as pitot, GPS further clogs the bandwidth resulting in very slow data rates form RX to Radio.
I have made requests to chose the data rate for each telemetry item. For example, I would send Battery 1&2 voltage every 5 seconds since it never change much frequently.  Also GPS clock doesn't need to send information very often.

The response frequency feature you have created is a big step in the right direction, but it would be much better if we can chose it on individual telemetry items.

Guys that have RB35S with 8 frsky xact servos are really dead in the water. There is just not enough bandwidth.
Thank you guys!

SmartTommy, 2024-08-28:
> @SmartTommy Thanks for the clarification. From my other ticket attempts, you are aware that the RB35S already by itself occupies most of the telemetry bandwidth right? And adding any additional [...]

can set all sensor to 0.1S~ 1S, this is a appropriate.

TD/TW Receeiver bandwidth rate is high enough to transfer 24  Xact servos telemetry data

noamkrief, 2024-08-28:
@SmartTommy 
So set in the lua to 3-10 right?
Are you saying RB35S + 24 Xact servos works well without bandwidth congestion over the 900HZ?

SmartTommy, 2024-08-28:
> @SmartTommy So set in the lua to 3-10 right? Are you saying RB35S + 24 Xact servos works well without bandwidth congestion over the 900HZ?
> 
> The issue is like I said, there are about 20 built-in telemetry sensors to the RB35S. Those 20 by themselves seem to congest the bandwidth.

Yes, set 3~10,if you use TW/TD, you can set to 1~10

TD receivers can Use  RB35S+ 8 Xact servos works well congestion over the 900MHZ when every servo and RB35S set to (0.3s~1s)

TD receivers can Use  RB35S+ 16 Xact servos works well congestion over the 2.4GHZ when every servo and RB35S set to (0.1s~1s)

In fact, due to protocol limitations, only 16  Xact  servos are allowed to be connected to a single FBUS/S.PORT line

noamkrief, 2024-08-28:
@SmartTommy 
So it seems TW which uses only 2.4GHZ has more bandwidth than TD which uses 900 for telemetry.
Is there a way to force a TD to use 2.4 for telemetry to get extra bandwidth?

SmartTommy, 2024-08-28:
> @SmartTommy So it seems TW which uses only 2.4GHZ has more bandwidth than TD which uses 900 for telemetry. Is there a way to force a TD to use 2.4 for telemetry to get extra bandwidth?

TD receiver send telemetry data use both 2.4G and 900M, At this point, the bandwidth is sufficient to transmit all sensor data in a timely manner， but after the plane flew far away(out the range of 2.4G), only 900M used.  

so If you are only flying close(within the range of 2.4G), the data bandwidth is sufficient for your use.

noamkrief, 2024-08-28:
@SmartTommy 
Thank you for the great communication.
Last question.
So when the RB35S is all alone without any other modules such as GPS, pitot, turbine ECU, the refresh rate for something like P.Angle, R.Angle seems quite nice.
As I move the RB35S, I can see P.Angle and R.Angle move very smoothly in the telemetry screen.

As soon as I plug in the GPS, pitot, the P.Angle and R.Angle response becomes very slow and sluggish.

Is that normal?

Because P.Angle and R.Angle is so slow, I don't get quality logs in the CSV export. I get new value every 0.5 seconds instead of every 0.1

closed 2024-09-13 by SmartTommy