FrSkyRC/RF_Feedback_Group #9: 2.4GHz signal NOT lost when rx turned off with tandem rx

Jedsters, 2024-08-26:
I am using a TD SR10 with the latest firmware and notice that sometimes when I power off the rx that the tx still thinks there is a signal. 
This means that if I try to turn the tx off I get the receiver still connected warning and have to accept that in order to turn the tx off. 

This photo shows at top right that the 2.4GHz signal is still full strength with 100db RSSI although the VFR 2.4G shows red, while the 900M signal is lost and red while display shows 

[image]

The telemetry page shows similar 

[image]

Thanks

RealTadango, 2024-08-26:
This always happens when i change a setting on the RX (TD series). It never happened when using the RX normally.

SmartTommy, 2024-08-29:
> I am using a TD SR10 with the latest firmware and notice that sometimes when I power off the rx that the tx still thinks there is a signal. This means that if I try to turn the tx off I get the [...]

Hi, could you tell us your Radio model（X20Pro,X20...）,software version, or could you show us the "info" page, so that we can reproduce this problem under the same conditions.