FrSkyRC/RF_Feedback_Group #39: Sensor Lost/Telemetry (TD-ISRM Pro 1.1.14 and Ethos 1.5.14)

Alex00101, 2024-08-27:
I do believe this is ISRM firmware issue rather than Ethos issue), in any case I'm reporting here as I'm not sure if there is a better register.
Updated my X20ProAW with TD-ISRM Pro 1.1.14 and Ethos 1.5.14. On two models where I tested the update the symptoms are consistent. Note that both models have Archer Plus stab receivers (SR8 and SR6 Mini) updated to the latest FW (v1.0.11), new stabs lua 3.0 installed, both receivers reset and recalibrated as per instructions. Both in ACCESS mode.

- Shortly after powering up and connection established, started getting "Sensor lost" messages.
- Went to model telemetry screen and all sensors except the basic (RSSI, RX Batt) - i.e. stab related, such as AccZ, P.angle are red.
- Deleted and rediscovered the sensors and only got 2 or 3 basic, nothing stab. specific.
- Tried to open the rx in the lua stab 3.0 script (thought I might need to reset it) - pressing Open caused nothing, i.e the script could not read the rx.
- cleared the rx in the FR menu, went through re-bind, got all sensors white and active initially but after a short while they all went red and I started getting "Sensor lost" messages again.
-------
Downgraded TD-ISRM to 1.1.13 and all symptoms are gone.

Once again I believe this is a serious ISRM firmware issue but this is the only place to raise it that I'm aware.

SmartTommy, 2024-08-28:
Thank you for your test, will fix it asap, please try the new TD-ISRMPro_1.1.15_20240828_Beta.frsk firmware

WhiskeyGolf, 2024-10-23:
I have the same problem with the TD-ISRM1.1.14 on Etho 1.5.18 firmware on the x20pro. In addition, sensor conflict warnings still occur like in the old firmware. Is there any news here or should I go back to 1.1.13?
Thank You

Alex00101, 2024-10-23:
I'm still on 1.1.13 awaiting any fix

SmartTommy, 2024-10-24:
We are currently modifying some features, which may take some time. During this period, you can use version 1.1.3， we will complete the update as soon as possible.

xalm17, 2024-11-15:
> We are currently modifying some features, which may take some time. During this period, you can use version 1.1.3， we will complete the update as soon as possible.

Hi I move to 1.1.14 with no issues so far, should I update back to 1.1.13, or it´s safe to be in the 14?

SmartTommy, 2024-11-16:
> > We are currently modifying some features, which may take some time. During this period, you can use version 1.1.3， we will complete the update as soon as possible.
> 
> Hi I moved to 1.1.14 with no issues so far, should I update back to 1.1.13, or it´s safe to be in the 14?

The 1.1.14 unable to connect properly when working in 900M Access mode, If you do not use this mode and do not use the 3.0.0 new features, you can continue to use the current version

Alex00101, 2024-11-16:
> The 1.1.14 unable to connect properly when working in 900M Access mode, If you do not use this mode and do not use the 3.0.0 new features, you can continue to use the current version

@SmartTommy may I please remind that original issue was that 1.1.14 was not working with normal Archer Plus stab receivers, not 900M.( sensor lost etc). You've just sent 1.1.15 beta that seemed to be OK on the bench but never made it to the official.
Would be nice to see some clarity in terms of when and what is going to be fixed in which version. Especially when each update requires full reset recalibrate resetup and often new lua. Please.