FrSkyRC/RF_Feedback_Group #40: Remote NFC with RB35S drops power

walrath596, 2024-08-29:
I have the RB35S and the new Remote NFC switch. ALL seems to be working correctly, however, a few times now the unit has shut off for no know reason and the only way to turn it back on is by un plugging and plugging the battery back in, or using the magnetic key i think too. Simply using the radio switch programmed to turn on the NFC switch does nothing. 

This could of course be catastrophic if it happened during flight.

SmartTommy, 2024-08-29:
Hi @walrath596 , do you mean you can turn on the Power with the magnetic key after the unit closed for unknown reasons ?

Can this phenomenon be repeated？

walrath596, 2024-12-23:
> I modified the code to recognize a unique ID to prevent interference, please update the ethos to 1.6.0 and use the new remoteNFC firmware, the new code alse change the logic to turn on/off the power:
>
> -100% ----> turn off
> +100%----> turn on
> others----> no change
>
> after upgrade the new firmware, you need bind your NFC to the CARD, and this CARD will the only KEY to turn on/off your device.
>
> RemoteNFC_20241219.zip
>
> If there are any issues, please let me know

Hello,
I will do that, but what is the “Card”?

SmartTommy, 2024-12-24:
> […](#)

“card" == "magnetic key"
please refer to :https://www.frsky-rc.com/remotenfc-panel/

walrath596, 2024-12-26:
How do you bind the Tag Key to the switch? There is no information on how to do that. Only on how to bind the switch. Thank you Sent from my iPhone

SmartTommy, 2024-12-27:
Press the "bind" button, then quickly bring your magnetic key close to your switch until it shows successful binding. If it fails, please repeat it
Please ensure that your ethos is version 1.6.0

walrath596, 2025-01-04:
I am not having any luck getting the tag key to work. My Remote switch bind successfully, and works remotely, but I can’t get the tag key to work. Can you please send a video on how you make it work. Thank you Sent from my iPhone

SmartTommy, 2025-01-04:
sorry, the new function need new ethos "device config" version, i will send you when finish building

bob01, 2025-01-18:
Using v1.6.0 -
Tried versions RemoteNFC_20241219.frsk (from this page) and RemoteNFC_20241109.frsk (latest on fr-sky.com)

Tried binging the NFC fob/card/dongle several times - cant get it to work. RF switch works (chan 12), dongle does not.

Please advise - a video demo would be great

SmartTommy, 2025-01-20:
> Using v1.6.0 - Tried versions RemoteNFC_20241219.frsk (from this page) and RemoteNFC_20241109.frsk (latest on fr-sky.com)
> 
> Tried binging the NFC fob/card/dongle several times - cant get it to work. RF switch works (chan 12), dongle does not.
> 
> Please advise - a video demo would be great

Sorry, The new firmware need new config tools witch will released with the 1.6.1

bob01, 2025-02-05:
Working as expected with 1.6.1 - thanks!

[image]

closed 2025-02-12 by SmartTommy