FrSkyRC/RF_Feedback_Group #41: accZ showing -1G. Should be +1G while sitting on the ground

noamkrief, 2024-08-29:
This is very basic.
I think it applies to all (S) receivers.
Thanks

doppleganger53, 2024-08-29:
If this is added it should be a non-default option to switch the sign, existing user model programming may be assuming current behavior.

noamkrief, 2024-09-04:
@doppleganger53 
I 100% understand your comment.
But should we keep an obvious error forever just because of that?

Like - if for the first 3 years of ethos, RSSI2.4 was mispelled to RSSSI2.4  
Are we now forced to live with that forever since some people may have programmed against that mispelled name?

One method is to fix it, and have the radio alert that this has changed and have the user acknowledge it after a firmware update that fixes it.
"ALERT! AccZ value is flipped in this firmware and future firmwares. Press OK to contunue"

SmartTommy, 2025-02-05:
firmware 3.0.x change -1 to 1. closed

closed 2025-02-05 by SmartTommy