FrSkyRC/RF_Feedback_Group #42: TDR10 AIN2 very slow to respond

noamkrief, 2024-08-30:
Hello.
When you hook up some volts to AIN2 lets say 3V

Then when you disconnect the 3V in telemetry, it will wind down to 0V over about 1-2 seconds.
So 
3V
2.5V
2V
1.5V
1V
1.5V
0.5V
0V after 1-2 seconds.

So it's slow to react to change in volts.

mawzthefinn, 2024-09-13:
What does it do if the voltage simply changes.

Removing the input leaves a floating sensor pin, which may behave oddly and is not a good test of sensor response

noamkrief, 2024-09-13:
Yes, even a change is just as slow as described.
In my particular application I got from 2S volt which is about 7v down to 2volts.
And the change back and forth is very lazy and slow.

When I use FAS40ADV instead of the AIN2, the change of voltage is immediate!!! So that's how I am certain it is an issue with the AIN2

mawzthefinn, 2024-09-13:
Thanks, that's certainly a valid test (from 7->2V)