FrSkyRC/RF_Feedback_Group #43: R9 SX not working prporly

sigurdma, 2024-09-06:
Hello
X20pro latest all.. It happened after latest RF firmware update I suspect.
R9 SX is not working correct. Tried two of them, same result. Se picture.
Falling in and out on rssi continuously. The rssi bar at right top is also going up and down. Somebody tried the same?
(TD R10 is ok)
[image]

SmartTommy, 2024-09-07:
Thank you for your test, could you please test the new firmware?
[TD-ISRMPro_20240907_Beta.zip](https://github.com/user-attachments/files/16914177/TD-ISRMPro_20240907_Beta.zip)

mgsanna, 2024-09-07:
> Thank you for your test, could you please test the new firmware? [TD-ISRMPro_20240907_Beta.zip](https://github.com/user-attachments/files/16914177/TD-ISRMPro_20240907_Beta.zip)

I had the same problems with R9 Slim - OTA, solved with the new firmware of X20PRO 1.1.15
Thanks

sigurdma, 2024-09-07:
Hello

Now its OK. Tested ok with R9 SX (and also TD R10)

Regards Sigurd

Den 07.09.2024 09:38, skrev Marco Gualtiero Sanna:

closed 2024-09-12 by SmartTommy

Dsrenger, 2024-10-09:
Is this issue incorperated into the current firmware 1.1.14 I can find on www.frsky-rc.com?

SmartTommy, 2024-10-09:
> Is this issue incorperated into the current firmware 1.1.14 I can find on [www.frsky-rc.com](http://www.frsky-rc.com)?

The TD_ISRMPro_3.0.0 firmware fixed it.   you can find here:https://github.com/FrSkyRC/RF_Feedback_Group/releases/tag/Beta