FrSkyRC/RF_Feedback_Group #44: msp performance on tw range of receivers.

robthomson, 2024-09-09:
I have recently built a small tool to anaylise msp throughput to the rotorflight flight controller.

This tool is relatively simple.

For 30s it requests large msp datasets from the fbl; and then returns some stats around what the outcome was.

Stats in my case are as follows:

**ELRS:**
Total queries:            50
Succesfull queries: 50
Timed out queries:  0
Protocol retries:      4
Average query time: 0.6s

**FRSKY ARCHER:**
Total queries:            49
Succesfull queries: 49
Timed out queries:  0
Protocol retries:      4
Average query time: 0.61s

**FRSKY TW**
Total queries:            42
Succesfull queries: 42
Timed out queries:  0
Protocol retries:      14
Average query time: 0.71s

In all cases I am on a Tandem Pro - Ethos 1.5.15 and latest RF/RX firmwares.

Whats absolutely clear is that the TW protocol is lagging behind.  It has taken considerable effort and sacrifices within RF2ETHOS to make the TWRX receivers work due to the high amount of retries.  These stats clearly show the issue.

Functionally.. the TW receivers have fanstastic range and latency - but there is an issue that needs investigating on the delays / retransmits when sending sport data.  

It would be great if this could be looked into.

Rob

robthomson, 2024-09-13:
So intitial tests on the beta firmware on TW receivers and X20PRO

old firmwere:  
0.6s avg query time
retries  10

beta firmware
0.53 avg query time
retries 2

This initially seems positive.

robthomson, 2024-09-20:
So all up there is still something functionally wrong with the TWMX msp implimentation.

Typical example:

- connect.  send msp.  Its supper fast.
- wait 2 mins
- do same test.  all initial speed lost and very slow.

rburrow87, 2024-09-21:
Yes I've noticed the same issue with Twin in both v3 beta and the current release versions. Every now and then it'll connect and perform flawlessly, but more often than not, it ends up slower and with many more errors/retries in this test.