FrSkyRC/RF_Feedback_Group #45: Vspeed in telemetry making no sense

noamkrief, 2024-09-09:
Vspeed from my R35S makes no sense.
Maximum is 2.5m/s during my vertical climbs at nearly 100knots.
2.5m/s is around 500fpm. 
Going vertical at 100knots should be more like 10,000 f/m

Also when I descend, the values of Vspeed don't go to negative. They should.

[image]

noamkrief, 2024-09-09:
@bsongis-frsky 

Yes, I think it's an RF issue. Your Ethos is just displaying the data given by the sensor :) thank you for all the amazing hard work!!!!

Nicholas-Luoyi, 2024-09-10:
This should be a decimal issue in ETHOS. I have pushed the fix for this. @bsongis-frsky