FrSkyRC/RF_Feedback_Group #47: Beta v3 firmware update

Alex00101, 2024-09-13:
An important question is that for gyro enabled receivers does new firmware means taking them out if the plane and recalibrating?

Because I've just done that (take receiver out of the model, reset, calibrate, install back) for the latest version of ArcherPlus SRn firmware for all my affected receivers to be compatible with the last round of rx firmware updates and v3 lua, I hope this firmware version does not force me to do it all over again.

Thanks

SmartTommy, 2024-09-13:
The current SRn firmware can be used normally and does not require any updates if you don't need the new function of the V3

Alex00101, 2024-09-13:
Sorry but this is not the question I'm asking.

As I need the new functionality as especially because your 1.1.14 isrm is
buggy (confirmed by you)  mu question is

If I update to v 3 my SR receivers, will I have to take all these receivers
out to recalibrate.

Could you please kindly answer that

SmartTommy, 2024-09-13:
> Sorry but this is not the question I'm asking. As I need the new functionality as especially because your 1.1.14 isrm is buggy (confirmed by you) mu question is If I update to v 3 my SR receivers, will I have to take all these receivers out to recalibrate. Could you please kindly answer that
> […](#)

if you used ACCESS SRx Version 1.0.11, you can only upgrade your Tx to V3, they are compatible. otherwise，You need to update both the RX and TX to V3.0

Alex00101, 2024-09-13:
Thank you, if I update Archer Plus SR8 to v3, do I need to recalibrate it
(ie take it out from the model)

SmartTommy, 2024-09-13:
> Thank you, if I update Archer Plus SR8 to v3, do I need to recalibrate it (ie take it out from the model)
> […](#)

After upgrading the receiver, you need to rebind and turn on the Gyro again, check the parameter, In theory, there is no need for recalibration, but I cannot guarantee 100%

Alex00101, 2024-09-13:
Thanks that answers my question. May I please suggest FrSky - for future rx
firmware upgrades- takes particular care of the need to recalibrate the rx
after firmware upgrade and either provide a clear instruction when
calibration settings are erased and recalibration is needed or better
design the update process so that recalibration is NOT required.

That would be very appreciated and welcomed. Thanks

philippe-warnotte, 2024-11-04:
Is the firmware FW-RF-Modul-ISRM-Twin-Beta.v3.0.0 just a test version? Is it possible to fly with this version 3 without risk?
 Is the file: "TW-ISRM_3.0.0_Beta.frsk" compatible with TX Kavan V20?

SmartTommy, 2024-11-05:
> Is this firmware FW-RF-Modul-ISRM-Twin-Beta.v3.0.0 just a test version? Is it possible to fly with this version 3 without risk? Is the file: "TW-ISRM_3.0.0_Beta.frsk" compatible with TX Kavan V20?

it's compatible with TX Kavan V20

philippe-warnotte, 2024-11-05:
And Is it possible to fly with this version 3 without risk? Or waiting the official version...