FrSkyRC/RF_Feedback_Group #48: Twin Lite Pro - sensor names still ambiguous

BladeScraper-Designs, 2024-09-13:
X18SE
Ethos 1.5.15
Twin Lite Pro External RF Module - V3 Beta FW
TW Mini - V3 Beta FW (same as TW MX apparently?)

Sensor names from TW Lite Pro seemingly not fixed like they were in TD-ISRM-PRO for TW.  There are now two separate RSSI sensors as there should be, but they are only named "RSSI".  
[image]

Apparently this issue is not present on TD-ISRM-PRO, as the sensor names are distinct there:
[image]

Additionally - MSP telemetry communication on TW Lite Pro seem to not work at all, at least on V3 Beta FW (have not tried previous version).  Reportedly working fine with TD-ISRM-PRO, but with TW Lite Pro, telemetry works but Integration for Rotorflight does not work.  I have tested with AP SR8 and TW Mini, both on V3 Beta FW.  In both receivers, the integration times out and is unable to communictate, despite telemetry itself (e.g. Rotorflight telem sensors) working fine.  I have not tested with other FC units but imagine it will be the same.  
[image]

BladeScraper-Designs, 2024-09-13:
Additionally, when going in the "Info" page in Ethos, there is Telemetry lost.  This issue was supposedly fixed in TW Lite Pro FW 1.07 (2024-08-1), but it seems to be back in 3.0 Beta.

SmartTommy, 2024-09-24:
Hi ,could you send us the MSP tool? and how to use this tool?