FrSkyRC/RF_Feedback_Group #52: TWMX Fix please

robthomson, 2024-09-23:
When can we expect a fix for the abismal MSP performance on the tw receivers.

ARCHER ->  96 queries in 30s succeed.
TW -> 55 succeed.

I am rapily loosing interest in TW and considering switching to archer.

rburrow87, 2024-09-23:
Even more frustrating is sometimes you can power everything on and it works great, and other times (more often than not), it's very poor. I see this same behavior with both the v1.x release and the new v3 beta.

SmartTommy, 2024-09-24:
could you share us your msp tools?

BladeScraper-Designs, 2024-09-24:
> could you share us your msp tools?

It is built into the Rotorflight Ethos Integration app.  Here:
https://github.com/robthomson/RF2ETHOS/releases

You will need a flight controller flashed with Rotorflight in order to use it.  Any STM32F722 Betaflight FC can be flashed with Rotorflight.  Configure it for F.Port receiver, then you can try the Lua.

robthomson, 2024-09-24:
Correct.  You will need a remote flight controller.

This issue will affect all systems talking to a remote device over the rf link - essentially twmx is slow and has multiple timeouts.

SmartTommy, 2024-09-25:
> Correct. You will need a remote flight controller.
> 
> This issue will affect all systems talking to a remote device over the rf link - essentially twmx is slow and has multiple timeouts.

I have a RotorFlight2, how to config it and connect to the TWMX?

robthomson, 2024-09-25:
You will want to wire up using fport on the receiver, and connect up to one of the uarts on the flight controller.

@BladeScraper-Designs shouls be able to help with this! He works for Frsky usa

SmartTommy, 2024-09-26:
Hi @robthomson ，We have identified this issue and it will take some time to fix. We will complete it as soon as possible.

robthomson, 2024-09-26:
Great stuff.

Thanks for the update.

Rob