FrSkyRC/RF_Feedback_Group #54: RSSI value with RX8R-PRO ACCST V2

aviat40, 2024-10-18:
Hello,

With ETHOS on X20 / X20S / X10S / X12, tested on FCC or LBT, on ETHOS and OTX,  the RSSI value is very low.
Even if both are closed ( 20cm ), the value is about 80 and very fluctuating. 

In flight, the value drop below 35 and there is warning all the time.
RX et TX are updated. I have changed from internal to external antenna.. same issue. And also new antennas on the RX.

Both RX and TX has been removed from the A/C to test in a free environment.

I wondered if it is a Bug between the ISRM module

Thanks to help

BR
Thierry

aviat40, 2024-10-19:
Hi,
Tested at the same place with a SR6 latest firmware FCC... RSSI is OK
Tested with a second RX8R-PRO 2.1.11 FCC still same place ... RSSI drop

The problem is clearly in the RX8R-PRO firmware 

If you need more test. I can make a video and send via discord or any other PM.

Thierry

SmartTommy, 2024-10-19:
Do you have another RX8R-Pro receiver? could you conduct a comparative test using this two RX8R-Pro receivers?
I infer that there is a hardware malfunction with your receiver， we can verify this speculation if you have two RX8R-Pro receivers

aviat40, 2024-10-19:
Hi
As I said in my previous message, i have tested with a second RX8R-PRO and it is the same. 
So it is not a hardware malfunction.

KAID-YANG, 2024-10-24:
Hello，This situation is normal and does not affect the range. If you're concerned about the range, it's recommended to use Archer Plus, TD, or TW series receivers.

aviat40, 2024-10-26:
Do you mean that I should not believe in the RSSI value and I should try to fly if there is a range issue ?
Are you serious !!! In this case, send me an RF aircraft and I will test for you.
I already change the RX with and Archer One but I have 2 of RX8R-PRO RX that I should be able to use

aviat40, 2025-02-16:
Any news for any update to have RSSI value correct ?

Thanks