FrSkyRC/RF_Feedback_Group #56: RX Lockup (Access, LBT, Archer Plus RS, F.Port)

jasc76, 2024-10-22:
Hi

Sunday I had two lockups with an Archer RS connected via F.Port to a nexus FC. This lockups happened for about 0.6s and the channels would go to 0 and the throttle to -100
[image]

As you can see here in the X20 logs, VFR stayed high and it even logged the dropped RPM
[image]

I cant really say what firmware is on the receiver. Is there a way to read it ?!

SmartTommy, 2024-10-22:
Hi， could you repeat this fault phenomenon？

jasc76, 2024-10-22:
Not yet
Also it's kinda scary with a 700 class heli. I was lucky not to break anything and this gets expensive quick

jasc76, 2024-10-22:
This is kinda interesting. I was playing around with the log and right around the cases with lost rc channel data, the rssi drops, but not below the threshold and link is not lost, vfr stays high. 
The heli was max 100m away
[image]

KAID-YANG, 2025-01-26:
Hello, could you provide the version information for the receiver, transmitter, and the internal RF module? Could you also try to reproduce this issue?