FrSkyRC/RF_Feedback_Group #58: Flight Data Record Fails - Archer Plus GR6 & GR8

Smoothkd, 2024-10-25:
### Flight Data Record (FDR) Fails to function. Tested on Archer Plus GR6 and GR8.

* Transmitter: X18S
* Ethos version 1.15.15
* TD-ISRM: HW 1.0.1 FW: 2.2.7 (FCC)
* Archer Plus GR6 Firmware: v1.01.10
* Archer Plus GR8 Firmware: v1.0.5

**GR6 1.05 Firmware Symptoms:**
With Rx registered and bound. When FDR option is chosen for RX1, Dialogue box shows "Waiting for Rx" after short period new dialogue box shows "No Response from RX". No options but to exit.

**GR6 1.01.10 Firmware Symptoms:**
Updated firmware OTA to 1.01.10 successful. When FDR option is chosen for RX1, **_It Unbinds and Unregisters Rx._** Solid Green light goes to Red. **In order to re-register and bind, I had to change Owner registration ID from default to new set of characters!!?**

**GR8 v1.0.5 Firmware Symptoms:**
Same as GR6 with FW v1.0.5

**Screenshots:**

[image]
[image]
[image]

mawzthefinn, 2024-11-05:
This is fixed in 3.0.0 beta, so once that hits general release the problem will be resolved.

Smoothkd, 2024-11-22:
A bet unclear to me,  Version 3.0.0 Beta?  Are you suggesting ver 3.0 of receiver firmware.  
GR6 current released FW is at ver 1.01.10    
Please elaborate

SmartTommy, 2024-11-22:
> A bet unclear to me, Version 3.0.0 Beta? Are you suggesting ver 3.0 of receiver firmware. GR6 current released FW is at ver 1.01.10 Please elaborate

3.0.0 will be release at the end of month

SmartTommy, 2024-11-22:
> Many thanks. I'll look for it, but 3.0 Beta. Seemed odd to jump from 1.0x to 3.0
> 
> These products are advertised with FDR feature built-in. Some FrSky features, as listed, seem to be feature requests. What are the exit criteria for Beta?

After completing preliminary testing internally, we will release a beta version, mainly for customers who have requested new features from us At the same time, we will conduct comprehensive testing (mainly distance testing and actual flight testing), and when we complete these, we will officially release them.

Due to the different version numbers of our products in different series, some users often make mistakes and feel confused. Therefore, we have unified all versions to 3.0

SmartTommy, 2024-12-23:
3.0.1 fixed this.

closed 2024-12-23 by SmartTommy