FrSkyRC/RF_Feedback_Group #61: Throw restrictions in the TD SR10 V. 1.0.13.

Richrfl, 2024-11-04:
After upgrading to Ethos 1.5.18 and the new LUA 3.0.0, resetting, and recalibrating the RX box and Sticks, I found that in AUTO level, the stick controls are strongly restricted—something that never happened in previous versions. Maybe I am missing something, but nowhere in the RX manual does it indicate a throw restriction parameter. Help, please.

Richrfl, 2024-11-04:
It seems I cannot do that anymore because this morning I deleted the RX1, and therefore I have no access to the clear option. Oh, boy, what a mess I have! Please, do not ask me to factory reset the TX. (LOL)

bsongis-frsky, 2024-11-04:
Why not :)

Richrfl, 2024-11-04:
I hope you are kidding about resetting the TX to factory, considering resetting one RX caused such a mess, resetting the TX could be beyond imagination.

Richrfl, 2024-11-11:
@bsongis-frsky / @mawzthefinn  guys, is there any possibility that I can recover the TD SR10 to life? it is a little expensive part to trash it.

Richrfl, 2024-11-16:
Gents, something else about this issue. I bought another TD SR10, updated the FW, Calibrated but did not Reset it as suggested. Everything is working fine and I do not have the throw restriction that was the origin of this thread. I hope this helps you to fix whatever killed my original RX. I will contact FrSky and see if they can set that RX back to normal, unless you suggest otherwise or have a better idea.

SmartTommy, 2024-11-18:
> Gents, something else about this issue. I bought another TD SR10, updated the FW, Calibrated but did not Reset it as suggested. Everything is working fine and I do not have the throw restriction [...]

@Richrfl , could you try the new firmware 3.0.0_Beta, this new firmware will reset all the parameter

Richrfl, 2024-11-18:
@SmartTommy, thank you for your suggestion, but I cannot afford to beta test. I do not have an extra transmitter, nor extra RXs to risk losing either. Otherwise, I would be happy to do beta as I have a model that is intended to test new FWs before installing and using new releases in my active models. I rather wait for the 3.0.0 release.

closed 2024-12-23 by SmartTommy