FrSkyRC/RF_Feedback_Group #65: P.Angle and R.Angle constantly increasing

noamkrief, 2024-11-18:
Roll and Pitch is derived from XYZ accelerometers via basic math.
If this math is done on ethos - it's an ethos issue. 
If it's done on the RB35S, then this needs to be moved to RF bugs.

RB35S
https://www.youtube.com/shorts/L8ksRgLoe4Q

xet303, 2024-11-19:
Same here with my new RB 25S. 
Checked in Basic mode only. 
Same with Ethos 1.5.18 & 3.0  & with latest firmware & RB tools.

RB 25S vidéo :
https://lesdesjardins.org/Nextcloud/index.php/s/Yc3n77HzeKsr8cF

SmartTommy, 2024-11-19:
@xet303 @noamkrief ，did your RB25S/35S work normally before？

xet303, 2024-11-19:
I can't say because it's brand new and this is the first time I've tried to get it to work.

noamkrief, 2024-11-19:
@SmartTommy 
Yes. It worked normally before.

SmartTommy, 2024-11-19:
> I can't say because it's brand new and this is the first time I've tried to get it to work.

could you try to do 6-axis calibration?

xet303, 2024-11-19:
I tried the 6-axis calibration several times & re flashed several times...always the same thing!

SmartTommy, 2024-11-19:
> @SmartTommy Yes. It worked normally before.

did you mean that they used to work normally, but now suddenly they don't work anymore? have you flash the firmware?

xet303, 2024-11-19:
> > @SmartTommy Yes. It worked normally before.
> 
> did you mean that they used to work normally, but now suddenly they don't work anymore? have you flash the firmware?

Yes, your question is relevant. The first thing I did before anything else was to update the RB.

noamkrief, 2024-11-19:
@SmartTommy 
Let me make this situation a little more STRANGE for you :)

I have been on 1.5.18 for over 2 weeks now and 3.0 LUA for at least 1 month.
I log ALL my flights. 
Flight from 1 week ago showing NORMAL situation.
Look at this 5 minute flight. It starts with Roll Angle of 0 deg and ends at 0 deg:

[image]

Now look, I just botted up my radio and RX and started recording for 10 seconds:

[image]

SmartTommy, 2024-11-19:
i test the firmware RB35S_v20240511.frsk again, works normal, Perhaps we need to test the hardware in order to identify the issue

noamkrief, 2024-11-19:
@SmartTommy 

[image]

[image]

noamkrief, 2024-11-19:
@SmartTommy 
If you want to remote into my laptop, I have the STK tool and you can troubleshoot the RB35S.
I can be available in exactly 3.5 hours from now. 10pm Pacific USA time

SmartTommy, 2024-11-19:
> @SmartTommy If you want to remote into my laptop, I have the STK tool and you can troubleshoot the RB35S. I can be available in exactly 3.5 hours from now. 10pm Pacific USA time

i will send you a sepcial firmware to find the problem

noamkrief, 2024-11-19:
@SmartTommy when can I expect this test firmware?

SmartTommy, 2024-11-20:
> @SmartTommy when can I expect this test firmware?

please test the new firmware, I'm not sure if this firmware can solve the problem
[RB35S_Test_30241120.zip](https://github.com/user-attachments/files/17823399/RB35S_Test_30241120.zip)

noamkrief, 2024-11-20:
@SmartTommy 
I re-flashed the firmware today and the problem immediately went away.
So maybe I should wait to see if the problem will re-occur and then try this test firmware.

SmartTommy, 2024-11-20:
> I tried the 6-axis calibration several times & re flashed several times...always the same thing!

@xet303 ,Your RB25S is still not working, is it?

noamkrief, 2024-11-20:
@SmartTommy 
I have a 2nd RB35S that just started having this problem.
using the test firmware you uploaded.

It's still doing it.

noamkrief, 2025-02-11:
nope! on the ethos side Bertrand does a great job resolving issues.
This RF side is crickets.

frskyrc-ben, 2025-03-27:
I was assisting a customer who reported some trouble with RB25S and this issue came up.

Here are the videos they shared

https://github.com/user-attachments/assets/d5f69df3-d56a-4c0b-8087-34bbaf0072d9

https://github.com/user-attachments/assets/9db092d0-7083-4529-8bc8-874a6ca4d27e

I verified with the user that they were doing all of the 6axis calibration and self-check function correctly. The unit was mounted correctly with the correct mounting type orientation set. The user also tried reflashing latest firmware 20240511 and with the results above.

https://www.dropbox.com/scl/fi/5qtolx7pdnpsyjvccp2k9/RB25S-Setup.mp4?rlkey=ypnqkd916hmfdf5whkhukgwy2&e=1&st=93wsm7zz&dl=0

My bench-test unit (early production) has little to no drift (+/- 0.01*) however this customer's unit seems to have significant drift.

Ultimately we will be replacing the unit for the customer and my personal opinion is a suspicion of bad or faulty gyro chip.

SmartTommy, 2025-03-28:
> I was assisting a customer who reported some trouble with RB25S and this issue came up.
> 
> Here are the videos they shared
> 
>  Messenger_creation_F8865F65-A1A3-4D52-9FD5-08DBA4DA144B.mp4 
>  Messenger_creation_377C4E5D-8D7A-406E-9383-C9C4F83AA0E2.mp4 
> I verified with the user that they were doing all of the 6axis calibration and self-check function correctly. The unit was mounted correctly with the correct mounting type orientation set. The user also tried reflashing latest firmware 20240511 and with the results above.
> 
> https://www.dropbox.com/scl/fi/5qtolx7pdnpsyjvccp2k9/RB25S-Setup.mp4?rlkey=ypnqkd916hmfdf5whkhukgwy2&e=1&st=93wsm7zz&dl=0
> 
> My bench-test unit (early production) has little to no drift (+/- 0.01*) however this customer's unit seems to have significant drift.
> 
> Ultimately we will be replacing the unit for the customer and my personal opinion is a suspicion of bad or faulty gyro chip.

Hi ben, 
Based on your description and the phenomenon in the video, I believe that the RB25S has lost calibration data.
Please try recalibrating of RB25S Gyros and Accelerometer

frskyrc-ben, 2025-03-28:
> Hi ben, Based on your description and the phenomenon in the video, I believe that the RB25S has lost calibration data. Please try recalibrating of RB25S Gyros and Accelerometer

It was just calibrated moments before the video with the phenomenon

The calibration was done as shown in this video : https://www.dropbox.com/scl/fi/5qtolx7pdnpsyjvccp2k9/RB25S-Setup.mp4?rlkey=ypnqkd916hmfdf5whkhukgwy2&e=1&st=93wsm7zz&dl=0

noamkrief, 2025-08-02:
@SmartTommy 
@bsongis-frsky 

This issue is back with the latest 3.0.6 FW of RB35S