FrSkyRC/RF_Feedback_Group #66: RB25S and Xact Servo incompatibility.

BJR981S, 2024-11-22:
Warning - RB25S not compatible with Xact servos.

Building my new test aircraft as previously advised I have found this issue with the RB25S.
The servos once configured will operate on the RB25S but unreliably.

Once configured, means you cannot configure them on the RB25S you need to use a TD or RW Rx to do that.

The process to configure an Xact servo is to capture it with add sensors in the telemetry menu.
You then select one of the telemetry lines associated with that servo and select device config.

With Device config you can then set the Phy ID and the Software group ID. and the Servo Channel number. Then save to flash.

You have to do this one servo at a time as the all default to the same channel 1 settings. (After you upgrade the FM)

The device config on the RB25S will not display the settings and they are greyed out. So no setup possible.

Normally on startup the Xact Servos are interrogated and there settings become available. (TD/TW Rxs)

The does not happen on the RB25S.

This warning relates to, as these settings are not loaded, there is a probability that the Servo will not react to its previously programmed centre  or end points correctly.

These can randomly wander. 

I will lodge a bug fix request for Ethos.

This was tested and performed multiple time. The centre wondering occurred on 2 of the 12 servos in the configuration. While you can see the telemetry the telemetry is not interogated and does not display correctly when selected. Moving the servo from the RB25S to the TW 12S the servo will have the correct centre point. Moving it back to the RB25S the centre point wanders. The servo is interrogated via FBUS signalling and the servo moves to the centre point on power up if the transmitter is signalling at centre. When the wander occurs. (Always one the same servo so about 8% of them) the wander happens very slowly. Subsequent operation of the servo from 1500us to 2000us or 1000us has a different range by the amount of the wander. By connecting the servo via PWM (Change FBUS port to PWM) it operates correctly. Issue only occurs with some servos when FBUS connected and only on the RB25S. I also had 1 of the 12 fail totally and it was dangerous. I badly burnt 2 fingers. On power up it started melting the servo connection cable. I burnt my fingers pulling the cable from the RB25S 2nd degree burn. The servo was operating OK for prior to that power up. I will document other issues that I have had with this build including issues with the TWMX mini. after more testing. I will go back to evaluating the RB25S and the RB35S I will capture the servo wander on video. The program centre may well be set on the servo at configuration time but the actual arm position will be at what the FBUS signal for that servo physical ID indicates. When the FBUS initialises with the Servo it handshakes to firstly determine that it is FBUS capable and what the channel, physical ID, software ID information is set at and initialise its telemetry communications. These small percentage of servos somehow get the position wrong and wander. I suspect a bios controlled FW Bug on the RB 25S. Suspect signal Timing. To correct your last point The telemetry signalling to the RB25S operates correctly. The telemetry from the RB25S does not, select one of the Servos where the servo telemetry says Servo X OK and try and configure the servo. The selection is greyed out and the information it provides is incorrect. So the FBUS Telemetry is only working one way. Not both ways as it should.

SmartTommy, 2024-11-22:
Hi, I didn't understand the problem you described. Can you take a video for us?

BJR981S, 2024-11-22:
The device config on the RB25S will not display the correct settings when viewed via the Telemetry entry they are greyed out and can not be configured.

The telemetry functionality is only working from Xact Servo to RB25S. The RB25S to Xact servo Telemetry control does not function. The FBUS is only working one way.  

Start with this. It is the root of the issue.

I have moved the servos to my model and replaced the RB25S with a TW12S. I have 4 new RB25Ss and 30 Xact servos on the way. Will retest when they arrive and video.

SmartTommy, 2024-11-22:
> The device config on the RB25S will not display the correct settings when viewed via the Telemetry entry they are greyed out and can not be configured.
>
> The telemetry functionality is only working [...]

will test and reply you asap, thanks for your test

SmartTommy, 2024-11-22:
[RB25S_20241122_test.zip](https://github.com/user-attachments/files/17868368/RB25S_20241122_test.zip)

Hi, Please test the new firmware

BJR981S, 2024-11-23:
I have conducted tests on this firmware. It appears to have solved the Bidirectional connection via FBUS. I will do full regression testing but it will take some time. I would recommend that you release it to the public. I will also test all of the Redundancy bus models and advise if others share this issue.

BJR981S, 2024-11-24:
Hi,

As advised the RB25S firmware tested OK.

I have now tested the RB35S.

It has the same issue as the RB25s had before the FW update. Single direct FBUS telemetry to Xact Servos and can’t be configured.

Can you provide a RB35S firmware update for me to test please.

Cheers Brian

SmartTommy, 2024-11-25:
> RB35S has same problem. Can I get new RB35S FW to test please.

SmartTommy, 2024-11-25:
> RB35S has same problem. Can I get new RB35S FW to test please.

[RB35S_20241125_Test.zip](https://github.com/user-attachments/files/17895797/RB35S_20241125_Test.zip)

BJR981S, 2024-11-25:
Testing done. New FW has corrected the issue. I will close this now.

BJR981S, 2024-11-25:
Tested update oK. All good to publish.

Thanks.

closed 2024-12-05 by BJR981S

doppleganger53, 2025-03-05:
I assume the RB35 non-S version will be getting these fixes as well? Any update on the official release?

SmartTommy, 2025-03-06:
> I assume the RB35 non-S version will be getting these fixes as well? Any update on the official release?

you can  flash the RB35S firmware. The software can recognize your hardware

doppleganger53, 2025-03-13:
> > I assume the RB35 non-S version will be getting these fixes as well? Any update on the official release?
> 
> you can flash the RB35S firmware. The software can recognize your hardware

That's fine for testing, but I'd prefer to fly on a fully QA'd release. The general public should get these fixes on all impacted RB units with an official release through Ethos suite though?