FrSkyRC/RF_Feedback_Group #67: ISRM 3.0.0 - RxBatt missing on ACCST, ACCESS

gernotw, 2024-12-04:
with TW X14 Ethos 1.6.0 RC2 (+RC4) and ISRM v3.0.0 the receiver voltage seems to be missing in combination with a X4R with uni firmware V2-EU rev.83. Also with R6 access (latest firmware). With TW Mini receiver RxBatt is shown.
With TW-IRSM v1.0.7 RxBatt is transmitted

SmartTommy, 2024-12-04:
> with TW X14 Ethos 1.6.0 RC2 (+RC4) and ISRM v3.0.0 the receiver voltage seems to be missing in combination with a X4R with uni firmware V2-EU rev.83. Also with R6 access (latest firmware). With TW Mini receiver RxBatt is shown. With TW-IRSM v1.0.7 RxBatt is transmitted

what's the ratio you used with the ADC2?

gernotw, 2024-12-04:
on a R4X ADC2 is floating at 18 - 21V (RxBatt is 4.86V). I didn't use ADC2

SmartTommy, 2024-12-13:
Thank you for your test, fixed it:
[TW-ISRM_3.0.1_RC1.zip](https://github.com/user-attachments/files/18122553/TW-ISRM_3.0.1_RC1.zip)

gernotw, 2024-12-13:
Perfect! On X14, 1.6.0RC7, V3.0.1 is transmitting RxBatt with X4R uni V2-EU rev.83
Thank you!

mawzthefinn, 2024-12-13:
This new firmware worked correctly with my Archer R4 for RxBatt.

It's still not respecting the Owner Registration ID properly, needed to re-register when moving receiver bind from my X20 & X20 Pro (both on RF 3.0.0 and same Owner Registration ID)

gernotw, 2024-12-17:
On X14, 1.6.0RC9, V3.0.1 is transmitting RxBatt with X4R uni V2-EU rev.83, but the value is frozen (in my case to 4.03V with one lipo on X4R)

SmartTommy, 2024-12-18:
> On X14, 1.6.0RC9, V3.0.1 is transmitting RxBatt with X4R uni V2-EU rev.83, but the value is frozen (in my case to 4.03V with one lipo on X4R)

on my x14 1.6.0RC7 V3.0.1, RxBatt shows correctly when X8R power changes. do you have X4R or X8R with frsky firmware?

gernotw, 2024-12-19:
> on my x14 1.6.0RC7 V3.0.1, RxBatt shows correctly when X8R power changes

on x14 1.6.0RC10 V3.0.1 RxBatt is shown correctly on X4R with FrSky firmware. With X4R uni V2-EU rev.83 RxBatt is frozen

mawzthefinn, 2024-12-19:
Congrats, you've found a Uni-FW bug. You should report that to Mike Blandford, creator of Uni-FW

gernotw, 2024-12-19:
From MikeHRC I got just the hint, that 4V is the lower limit of power supply for X4R! Using 5V telemetry / RxBatt is ok!! The issue can be closed!

closed 2024-12-20 by SmartTommy