FrSkyRC/RF_Feedback_Group #69: RB25S and RB35S S-Port has same problem a FBUS before fix

BJR981S, 2024-12-05:
RB25S and RB35S FBS unidirectional issue has been fixed by test FW update. Further testing has revealed that the same issue is present with configuring the channel to S-port. The telemetry is received but you are unable to configure the S-Port device from the Telemetry Page or the LUA script. This is identical to the use that has been resolved with FBUS.

Issue found during testing of a Neuron 8A SBEC.

BJR981S, 2024-12-23:
Please provide firmware update to me so I an confirm if this is fixed or not.

BJR981S, 2024-12-23:
Please provide the FW update so I can confirm that this is indeed fixed.

Closed #69<https://github.com/FrSkyRC/RF_Feedback_Group/issues/69> as completed.

SmartTommy, 2024-12-24:
[RB35S 3.0.0.3.zip](https://github.com/user-attachments/files/18236235/RB35S.3.0.0.3.zip)
[RB35S_3.0.1_RC1.zip](https://github.com/user-attachments/files/18236237/RB35S_3.0.1_RC1.zip)

SmartTommy, 2024-12-24:
we test the Neuron BEC 8A with the S.PORT connect to the RB35S and it works well both telemetry and config the BEC, this new firmware is not a release version,only for test
[image]

BJR981S, 2024-12-24:
Thank you for the RB35S Firmware. Can I have the RB25S firmware please. I am about to install it in a model.

xxx.jpg (view on web)<https://github.com/user-attachments/assets/6c0b13b6-87e5-4ac0-b10d-1d7de288b8e5>

SmartTommy, 2024-12-24:
> Referenc

[RB25S_3.0.1_RC1.zip](https://github.com/user-attachments/files/18236845/RB25S_3.0.1_RC1.zip)

note: both the R35/25 firmwares are only test version

BJR981S, 2024-12-24:
> > Referenc
> 
> [RB25S_3.0.1_RC1.zip](https://github.com/user-attachments/files/18236845/RB25S_3.0.1_RC1.zip)
> 
> note: both the R35/25 firmwares are only test version

Thanks. can I have the LUA as well please. I will do consolidated testing of the S-Port and FBUS fixes. Hopefully the FBUS fix is including in these test Firmware files as well?

BJR981S, 2024-12-24:
> > Referenc
> 
> [RB25S_3.0.1_RC1.zip](https://github.com/user-attachments/files/18236845/RB25S_3.0.1_RC1.zip)
> 
> note: both the R35/25 firmwares are only test version

I have tested the RB25S FW you provided. The test failed. I am testing from.

TX X20Pro AW. version 1.6.0 
RX TWMX Version 3.0.1 
RB25S Version 3.0.1 RC1

The test showed there is now an anomaly with the Xact Servo FBUS Telemetry. Its configuration ability from the telemetry menu still works. But there is a faulty telemetry input. It would normally say Servo1 OK. It is now a blank red line. Bt it still allows you to select and configure servo 1. Not sure why it is displaying incorrectly.

I tested with S-Port Connection to an 8M SBEC.The telemetry is visible but it cant be configured from the telemetry entry. This is the initial fault and is still evident in the test FW.

[image]

BJR981S, 2024-12-24:
also please reopen this case.

SmartTommy, 2024-12-25:
> > > Referenc
> > 
> > 
> > [RB25S_3.0.1_RC1.zip](https://github.com/user-attachments/files/18236845/RB25S_3.0.1_RC1.zip)
> > note: both the R35/25 firmwares are only test version
> 
> I have tested the RB25S FW you provided. The test failed. I am testing from.
> 
> TX X20Pro AW. version 1.6.0 RX TWMX Version 3.0.1 RB25S Version 3.0.1 RC1
> 
> The test showed there is now an anomaly with the Xact Servo FBUS Telemetry. Its configuration ability from the telemetry menu still works. But there is a faulty telemetry input. It would normally say Servo1 OK. It is now a blank red line. But it still allows you to select and configure servo 1. Not sure why it is displaying incorrectly.
> 
> I tested with S-Port Connection to an 8A Neuron SBEC. The telemetry is visible but it cant be configured from the telemetry entry. This is the initial fault and is still evident in the test FW. I don't have the updated LUA so cant test from that.
> 
> ![IMG_5154](https://private-user-images.githubusercontent.com/155904547/398478831-5e429ae8-ecc5-4ca0-a1c2-187730b4fae2.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUwODU5MjIsIm5iZiI6MTczNTA4NTYyMiwicGF0aCI6Ii8xNTU5MDQ1NDcvMzk4NDc4ODMxLTVlNDI5YWU4LWVjYzUtNGNhMC1hMWMyLTE4NzczMGI0ZmFlMi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjVUMDAxMzQyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGM2ZWMxNzEzYjI0YmEyMjkyOTI4YTI4ZDA4NWY3MjgyNzFkZGE5Y2NlMTEwOWFkMWE1NWFhYzU0OGVhMjQ5OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.h6Vp17LnBfNgIgK78yWjOhoxrW5R14KCfsUXyg1AI8Y)

The 8A Neuron SBEC works well here, could you show us a video ?" It is now a blank red line" the next ethos verison will fix this

BJR981S, 2024-12-26:
Link to Video.

I will do further testing with Different radio RX and RB25S unit. All pre 1.6.0 in case 1.6.0 is doing something wrong.

SmartTommy, 2024-12-26:
> Link to Video.
> 
> I will do further testing with Different radio RX and RB25S unit. All pre 1.6.0 in case 1.6.0 is doing something wrong.
> 
> https://drive.google.com/file/d/1SxQWWUcwb4OthjaKMUGSluBtPeiWlM53/view?usp=share_link

could you try to config the SBEC ----> Deveice Config--->ESC--->SBEC ?

BJR981S, 2024-12-26:
I did that. It does not work either.

BJR981S, 2024-12-26:
Can you reopen this please.

BJR981S, 2024-12-26:
> > Link to Video.
> > I will do further testing with Different radio RX and RB25S unit. All pre 1.6.0 in case 1.6.0 is doing something wrong.
> > https://drive.google.com/file/d/1SxQWWUcwb4OthjaKMUGSluBtPeiWlM53/view?usp=share_link
> 
> could you try to config the SBEC ----> Deveice Config--->ESC--->SBEC ?

Double checked. Input greyed out.

I have backedTX back, and RB25S back to previous Release.
Rolled forward again and no difference. 

I will test your RB35S update now and let you know.

Confirming the red telemetry line with no detail that used to read Servo 1 OK is an issue with 1.6.0 not your fix.

KAID-YANG, 2024-12-27:
https://github.com/user-attachments/assets/a31edf21-83ec-4188-8490-9d07019a76bb

SmartTommy, 2024-12-27:
> Double checked. Input greyed out.
>
> I have backedTX back, and RB25S back to previous Release. Rolled forward again and no difference.
>
> I will test your RB35S update now and let you know.
>
> Confirming [...]

it works normal, as the video posted by @KAID-YANG ,

BJR981S, 2024-12-27:
> it works normal, as the video posted by @KAID-YANG ,

I do not know why his is working. I do not know what versions of FW etc he is using. I do know that he is using 2 RXs he only shows the configuration of one. I do know that he is using SBUS 16 Channel mode I use SBUS 24 Channel mode.

So they are some fundamental difference.

I suggest you ask him to use a new radio / RB25s and RX. I have no idea what other mods have bee done to his equipment.

I have now tested across multiple Radios, Multiple RXs and Multiple RB units both RB35s and RB25s.

The 8 Amp SBEC configuration is greyed out on all my tests. What FirmWare is he using on his 8 Amp SBEC? Mine is not updated it is release version. There is no update on the FrSky site. 

Does the LUA script you provided need to be used to get it to work?

BJR981S, 2024-12-27:
> normal.video_20241227104439.mp4

Can you please provide Ethos Version. RB25S version. TWMX Version.

Can you please create a new model only bind the TDMX (you have 2 bound) Configure for 24 Channel SBUS.

You have  red warning on your display. What is the warning? 

Thanks

SmartTommy, 2024-12-30:
1. ETHOS 1.6.0  version :[https://github.com/FrSkyRC/ETHOS-Feedback-Community/releases/tag/1.6.0](url)
2. install the RB25S LUA from the suite or use this:
[RB25_S.zip](https://github.com/user-attachments/files/18271808/RB25_S.zip)
3. TD_ISRMPro 3.0.1:[https://www.frsky-rc.com/tandem-x20-pro-aw](url)
4. TWMX 3.0.1: [https://www.frsky-rc.com/tw-mx/](url)
5. RB25S: [RB25S_3.0.1_RC1.zip](https://github.com/user-attachments/files/18236845/RB25S_3.0.1_RC1.zip)
6. Reg and bind the TWMX, set FBUS (you can set the sbus to 16 or 24 as you want)
7. Connect the TWMX FBUS to RB25S
8. config the the RB25S pin9 to S.PORT and then connect this pin to the SBEC 8A
9. find the telemetry BEC ---> configure

SmartTommy, 2024-12-30:
> Link to Video.
> 
> I will do further testing with Different radio RX and RB25S unit. All pre 1.6.0 in case 1.6.0 is doing something wrong.
> 
> https://drive.google.com/file/d/1SxQWWUcwb4OthjaKMUGSluBtPeiWlM53/view?usp=share_link

I have watched your videos multiple times. Can you take a photo for me to see your welding points( the wire connect to the TWMX pad)

BJR981S, 2024-12-30:
I have now tested with:

- 3 different Radios upgraded to 1.6.0.
- 3 Different RB25S units. All Upgraded to your test FW
- 6 different Rxs 2 x TWMX 4 x TWMX Mini. All upgraded to 3.0.1 FW.
- 1 RB35S
- Same 8 Amp SBEC in all tests only common component What FW are you running on yours? Mine is initial release FW

The procedure outlined above was followed in all cases. Including a reset of the RB units as per FrSky Steves advice.

No issues with soldering PADS all tests included an XACT servo running in FBUS mode and telemetry entry used to access configuration menu. So end to end bidirectional access tested.

Pictures and Video.

[image]
[image]
[image]

[ ](https://drive.google.com/file/d/1HnkRZozflP8unKN-AsHM9qP-LIhEw0Zj/view?usp=share_link)

I will try another SBEC that is the only common hardware in the testing. Also what FW are you using on your SBEC?

Cheers

SmartTommy, 2024-12-30:
> I have now tested with:
> 
> * 3 different Radios upgraded to 1.6.0.
> * 3 Different RB25S units. All Upgraded to your test FW
> * 6 different Rxs 2 x TWMX 4 x TWMX Mini. All upgraded to 3.0.1 FW.
> * 1 RB35S
> * Same 8 Amp SBEC in all tests only common component What FW are you running on yours? Mine is initial release FW
> 
> The procedure outlined above was followed in all cases. Including a reset of the RB units as per FrSky Steves advice.
> 
> No issues with soldering PADS all tests included an XACT servo running in FBUS mode and telemetry entry used to access configuration menu. So end to end bidirectional access tested.
> 
> Pictures and Video.
> 
> ![IMG_5177](https://private-user-images.githubusercontent.com/155904547/399232749-3bc9f079-5754-4f49-800e-e9b334d5c37e.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1NDUwMjQsIm5iZiI6MTczNTU0NDcyNCwicGF0aCI6Ii8xNTU5MDQ1NDcvMzk5MjMyNzQ5LTNiYzlmMDc5LTU3NTQtNGY0OS04MDBlLWU5YjMzNGQ1YzM3ZS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMzBUMDc0NTI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2Y1MjIxNDlkMzVmYzAyNWFkZTczODM4YmQzMTFhZWI0MDUxOWU4OWRlMjQ0NzQ2MGE3ZTIxNWU5OGFlN2E0ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.uJrEMhQBqevToLJFtOXRj9jdoRBpXuMKHppZazL3Sr4) ![IMG_5178](https://private-user-images.githubusercontent.com/155904547/399232776-77835e7d-968b-430a-9f65-170be65dd049.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1NDUwMjQsIm5iZiI6MTczNTU0NDcyNCwicGF0aCI6Ii8xNTU5MDQ1NDcvMzk5MjMyNzc2LTc3ODM1ZTdkLTk2OGItNDMwYS05ZjY1LTE3MGJlNjVkZDA0OS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMzBUMDc0NTI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTM0MmZhYjE4MTViZjUyZmRmYWY3NjM0ZDhiMWYyYzk4YTUyYmU2OTg3YmVhNmFiMzU1NTJiZmZkOGYyYTEyOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.KYWeWJJevt3NVHaX_wncmGwt5a1Hg2AgAcs-iU_039c) ![IMG_5179](https://private-user-images.githubusercontent.com/155904547/399232801-2ffca8b3-1ab6-4885-bba8-9982109b5cb9.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1NDUwMjQsIm5iZiI6MTczNTU0NDcyNCwicGF0aCI6Ii8xNTU5MDQ1NDcvMzk5MjMyODAxLTJmZmNhOGIzLTFhYjYtNDg4NS1iYmE4LTk5ODIxMDliNWNiOS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMzBUMDc0NTI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDFlMGQ3NmY4MTMzYTdlZGQxNjc0NDVhNDg3Nzk1YWVhMTliNzMyYmExYTJjODQ3YjJiZGJmNjliZWRkYTM1MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.bMYFzkbqhYZ8t0t4oBg6KBupAdgULMnC-GvG8hdJLB0)
> 
> https://drive.google.com/file/d/1HnkRZozflP8unKN-AsHM9qP-LIhEw0Zj/view?usp=share_link
> 
> I will try another SBEC that is the only common hardware in the testing. Also what FW are you using on your SBEC?
> 
> Cheers

my SBEC is a new one, there is no firmware update.   could you connect your sbec directly to your Rx? i want to check if will work when connect to rx directly.

BJR981S, 2024-12-30:
> my SBEC is a new one, there is no firmware update. could you connect your sbec directly to your Rx? i want to check if will work when connect to rx directly. 

Yes have done that many times. That is how I set the Voltage on first use.

Ok. I got to the bottom of the issue.

I connected one of the test RXs to a S-Port hub with the SBEC and a 5.0 Volt NiMh battery pack. I have a video if you are interested. This ruled out everything to do with the RB25S

I tested this, and the telemetry, and the configuration, of the SBEC worked fine.

As a further investigation I changed the PhyID and the AppID to their max values.

Then retested on the RB25S. It worked fine.

Now what I don't understand is that the telemetry from the RB25S for the SBEC on the S-port connection worked OK. Just not being able to configure was the issue.

If there was a conflict in the PhyID or AppID it should have either not worked at all or at least generated a sensor conflict. Is this an issue with 1.6.0?

What settings do you have on your SBEC?. The FW should support the full PhyID and AppID available to the SBEC.

I was using the default settings as shipped. The only thing changed was the Voltage setting when I first set it up on a normal TW Rx.

I could have tested multiple 8A SBECs and they would have all failed as they would have identical default IDs.

I will try the same with the RB35S and see if that now works as well after changing IDs.

I will let you know.

SmartTommy, 2024-12-30:
> Yes have done that many times. That is how I set the Voltage on first use.
>
> Ok. I got to the bottom of the issue.
>
> I connected one of the test RXs to a S-Port hub with the SBEC and a 5.0 Volt NiMh [...]

Sorry, I didn't understand what you meant,can you briefly explain it? did your SBEC can be configure when connect to the RB25?

BJR981S, 2024-12-30:
> Sorry, I didn't understand what you meant,can you briefly explain it? did your SBEC can be configure when connect to the RB25?

By configuring the AppID and The PhyID of the SBEC to the max configuration setting of these two it now works with the RB25S and can be configured via the RB25S.

The default IDs of a new SBEC do not work. Hence why I could not get it to work previously.

What are the PhyID and AppID of your SBEC?

SmartTommy, 2024-12-30:
> By configuring the AppID and The PhyID of the SBEC to the max configuration setting of these two it now works with the RB25S and can be configured via the RB25S.
>
> The default IDs of a new SBEC do [...]

PhyID: 0x10  Appld: 0x0E50

BJR981S, 2024-12-30:
> PhyID: 0x10 Appld: 0x0E50

Thanks, Will try those settings and check if they work. I am going to get another SBEC out of my inventory and set it up and check its default settings. I want to determine what the actual setting is that makes it fail.

This should assist you in changing the firmware to support all SBEC settings.

I will update you once done.

BJR981S, 2024-12-30:
> > What are the PhyID and AppID of your SBEC?
> 
> PhyID: 0x10 Appld: 0x0E50

OK set the settings to:

PhyID: 0x12 Appld: 0x0E52

This is the default setting of the SBEC  8A. It will then fail and you can test what the issue is with these configuration settings.

I don't have any spare 8A SBECs so I have ordered a few more. It will  take a few weeks to get here from China.. But I do have a few 14A SBECs so I will test them as well and let you know. I will also adjust the settings up and down by one configuration at a time and see of there are other configurations that fail.

I tested the RB35S and it worked fine on alternate settings as well.

Cheers

closed 2025-02-06 by SmartTommy