FrSkyRC/RF_Feedback_Group #70: Inconsistent Temp values reported on Xact Servos Telemetry.

BJR981S, 2024-12-10:
I have tested 12 Xact serves unmounted close to each other with no servo commands to move and no servo arms attached, and watched the temperature reported in the telemetry menu.

The temperature ranges from 24C from lowest sensor reading to 35C degrees on the highest. They all seem the same temperature to the touch. I will test with laser temp sensor as soon as it arrives.

Servos all configured identical except Channel Value Phy and AP IDs.

Servos tested at a range from 5v to 8.5V no change detected.

WhiskeyGolf, 2024-12-18:
It's a shame that no one has commented on this yet. I think I have the same problem. My servos show differences of up to 20°C from a standing start. In cold environments, this sometimes leads to negative values ​​on some servos, which are displayed by ETHOS with extreme values ​​of over 240°C.
[image]

[image]

[image]

FrSky-GX, 2024-12-19:
@BJR981S @WhiskeyGolf    Can you supply the type of the servos used？

BJR981S, 2024-12-19:
The servos I used to test are 6 x M5253H, 5 x M5252H, 1 x M5251H. 12 Xact servos on 3 FBUS connections to an RB25S with TWMX mini Rx.

WhiskeyGolf, 2024-12-19:
6x M 5252H ( Ail, Fl, Elev ) , 2x W5651H ( Steer, RUDD ) 

On 5 FBUS outs off a TW SR12

bsongis-frsky, 2024-12-19:
Perhaps it is difficult to fix this in the receiver?
I will add an offset inside Ethos soon

WhiskeyGolf, 2024-12-19:
Is it a hardware problem? Bad temperature sensors?

SmartTommy, 2024-12-19:
we are testing the new firmware, will release asap

BJR981S, 2024-12-19:
> Does this mean a new firmware for the servos? I'm a little worried that there will be updates for servos, as I operate many servos via Y-cables and don't really have access to the individual servo connectors. Is there a way to update the servos together or separately via the ID over an entire cable?

Firmware updates need to be done via the S-Port connector on the radio or an STK S-Port USB tool. If there is a FW update then I hope it is soon before I seal in my 12 Xacts into my new model. As an Aside did you program the FW updates into your Xacts with the 15/02/2024 Xact FW release?

BJR981S, 2024-12-19:
> we are testing the new firmware, will release asap

Thank you much appreciated. When FW update released I will test with newly acquired Laser Temp sensor.

BJR981S, 2024-12-19:
> Perhaps it is difficult to fix this in the receiver? I will add an offset inside Ethos soon

As the reported temperature varies from Servo to Servo and is inconsistent, will this offset be done per each individual servo?

BJR981S, 2024-12-19:
> Hello @BJR981S . I know that there are two options for updates. Since my servos are already installed and solder together, I can no longer go to the classic way of individual servos to updates because only Servorails for entire areas arrive at my receiver. I'm running on the last FW ( 2022-02-15 V2.0.1 )

Damn, thats a PITA. I have not sealed my servos in, and I use the FrSky S-Port interconnect modules rather than solder. So I guess my model build will just wait for a while. Yes my bad I typed 2024 rather than 2022. So more than 2 years with no update. I guess one is due. I currently have all the Xact brushless and coreless servos in my build supplies. 130 in total.

I will wait for a while to glue the 2 fuse halves together. (Upper /Lower)

[image]

WhiskeyGolf, 2024-12-19:
@BJR981S ,Could you please do me a favor and expose a servo to a cold environment and see if it goes above 240°C? Just to make sure we have the same problem

bsongis, 2024-12-19:
@WhiskeyGolf This problem is fixed already in 1.6.0 which should be released tomorrow.

BJR981S, 2024-12-19:
> @BJR981S ,Could you please do me a favor and expose a servo to a cold environment and see if it goes above 240°C? Just to make sure we have the same problem

What temp would I need to chill to? i guess I could put one in the freezer. its 35-40C here at the moment. I suspect that this is a negative sign error in the high bit of the telemetry data.

BJR981S, 2024-12-19:
> @WhiskeyGolf This problem is fixed already in 1.6.0 which should be released tomorrow.

If not released tomorrow what RC Candidate is the fix in. Just the latest?

WhiskeyGolf, 2024-12-19:
Yes, it would have to go into the negative. So if a servo is already showing too little at 30°C, you don't have to cool it down very much. Only if it's not a problem

WhiskeyGolf, 2024-12-19:
I've tested it outside again. After 0 degrees it got 255. That's good to know. At first I was worried that it could be my thrust tube

FrSky-GX, 2024-12-20:
@BJR981S @WhiskeyGolf   I suspect that there is a high probability that it is a servo software problem, and we will test it soon.