FrSkyRC/RF_Feedback_Group #71: RF-TW-ISRM V3.0.0 + Ethos 1.6RC6:  Alarming "Telemetry Lost" while staying in ETHOS's [Info] tool.

philippe-warnotte, 2024-12-11:
It seems that this is the same issue as TW-ISRM v1.0.6 fixed by V1.0.7.(See release Note)
Testing with Kavan V20 Twin + RX TWR8 and also TW-Sr8.
[image]
[image]

SmartTommy, 2024-12-12:
when enter "info" page, Tx will polling Rx1---->Rx2---->Rx3, if only connect one or two Rx, the Telemetry will lost when search the Rx not connected.

This phenomenon will not affect normal use, there is no need to worry

philippe-warnotte, 2024-12-12:
Thank you for your answer. I understand very well what you are explaining to me and I don't worry.
However, can you tell me why this phenomenon was declared as issue in version TW-ISRM 1.0.6 and 
was then fixed with version 1.0.7? Below, your release note:

*******************************Release Note******************************************
This Package is for TW-ISRM module of TWIN radios v1.0.7 firmware update. 
Version and Files: 
TW-ISRM_1.0.7.frsk                            
Upgrade file for the upgrade TW-ISRM module for TWIN XLite/XLite S/X14/X14S/V20.                                               
Release note 
Firmware Version: 【v1.0.7】
1. Fixed the issue of the repeat telemetry data of the same sensor source.
2. Fixed the issue of alarming "Telemetry Lost" while staying in ETHOS's [Info] tool.
----------------------------------------------------------------------------
Firmware Version: 【v1.0.6】
1. Added the support of configuration capability for the RemoteNFC's features. 
2. Added the hardware compatibility of Kavan V20 radio.
-----------------------------------------------------