FrSkyRC/RF_Feedback_Group #72: TD ISRM X20S 3.0.1 rc1 doesn't work for TD protocol

mgsanna, 2024-12-13:
X20S  - 1.6.0 RC7 Flex - TD ISRM 3.0.1 RC1 
TDSR12 3.0.1 RC1 - TDR18 3.0.1 RC1 - TDSR6 3.0.1 RC1

Same procedure, on radio : 1 registration, 2 binding 
radio recognise the receveir in registration and also in binding, and say connection ok, but no signal at all from receiver;
RSSI and VFR stay at 0, on reeiver the led stuck on green...

Flashing in OTA internal works as supposed but no signal at all

The same procedure and process is working as expected on my X20PRO and X20PROAW

How to solve?
ps ACCESS protocol works without problems on 3.01 RC1 X20S
ps2: no response from RX (in rx menu and option)
ps3 obviously no Telemetry at all
 reverting to 3.0. beta everything is working as supposed

aviat40, 2024-12-15:
I confirm, 
X20S 3.0.1 and TDR10 3.0.1
Binding is OK, servos are OK,
No RX options and no telemetry at all

MIKE-DAILEY, 2024-12-15:
Replicated here too.  Testing with 3.0.1. RC1 version dated 12/11/2024 4:48 PM

*Working OK*
X20 Pro,  X20 Pro AW ETHOS 1.6.0. RC8 OK with TD, TW and ACCESS. 
TW X-Lite ETHOS 1.6.0. RC8 OK 

*Not working* 
X20S ETHOS 1.6.0. RC8 TD R10 only green LED and no telemetry but channels work
X18SE ETHOS 1.6.0. RC8 TD R10 only green LED and no telemetry but channels work

SmartTommy, 2024-12-16:
[TD-ISRM_3.0.1_RC2.zip](https://github.com/user-attachments/files/18142988/TD-ISRM_3.0.1_RC2.zip)

Thank you for your test, fixed it.

MIKE-DAILEY, 2024-12-16:
TD-ISRM 3.0.1. RC2 test OK with X20S and TD R10 RX.  Will test on X18SE next.

mgsanna, 2024-12-16:
Tested on X20S ethos 1.6.0 RC8 and TD ISRM 3.0.1 RC2  work OK with almost every TANDEM RX  (TD R18 - SR12 - SR 6 -R10 and MX)

Thanks

closed 2024-12-16 by mgsanna