FrSkyRC/RF_Feedback_Group #73: Strange telemetry spikes with receivers with S.Port on PWM connection

glipski50, 2024-12-13:
X20R Ethos 1.6.0 RC 05 and ISRM 3.0.0
TW R10 FW 3.0.0
Sensor FLVSS most firmware actual

Not sure if it is a RF issue. If it is Ethos please move it to Github/Ethos_Feedback_Group

Using a TW R10 and a FLVSS telemetry data shows occasional spikes, positiv or negative which are not real values but obviously caused by corrupted data (or so). 

Obviously this behaviour can only be seen with reveivers that have the S.Port programmed to a PMW connection like TW R10 or TW R12. This behavior CANNOT be seen with receivers reveivers that have the S.Port connection on the top of the receiver like TW R8.

This behavior shows also up with XEs with Ethos 1.4.17 and Twin X Lite with Ethos 1.5.19. In both cases  RF firmware versions are prior to Beta 3.0.0.

See below screenshots and datalog. With this wrong data Min- or Max voltage on the tx display do not make sense and are very confusing.

[image]

[image]

[image]

HolgerJr, 2024-12-13:
I have the same thing with receiver Plus SR10+, Plus R12+, TD-R10.

Plus SR10+
[image]

[image]

[image]

Plus R12+
[image]

[image]

TD-R10
[image]

[image]

[image]

glipski50, 2024-12-13:
Same issues with RF version 3.0.1

[image]

glipski50, 2024-12-14:
Also with RF 3.0.1 those spikes show up

[image]

SmartTommy, 2024-12-16:
could you send us the log file?

Hartinger, 2024-12-16:
Today I observed exactly the same phenomenon for the Rx voltage from an Archer R8 pro receiver (FW 2.1.14). Tx was a TWIN X14S with ETHOS 1.5.19 and module FW 1.07 EU-LBT. So it seems that the problem does not only occur for sensors connected to the S.Port but also for Rx internal telemetry values. There is obviously a telemetry value missing in the log file.

[image]

glipski50, 2024-12-16:
> could you send us the log file? and which sensor you used（FLVSS or FLVSS ADV）?

It's the FLVSS.

[LiPo-Voltage-Log.zip](https://github.com/user-attachments/files/18149425/LiPo-Voltage-Log.zip)

SmartTommy, 2024-12-17:
> > could you send us the log file? and which sensor you used（FLVSS or FLVSS ADV）?
> 
> It's the FLVSS.
> 
> [LiPo-Voltage-Log.zip](https://github.com/user-attachments/files/18149425/LiPo-Voltage-Log.zip)

I spent a whole day without repeating this phenomenon. I have created a testing program, could you help me test it?

[TWR10_3.0.1_FLVSS_TEST.zip](https://github.com/user-attachments/files/18163399/TWR10_3.0.1_FLVSS_TEST.zip)

glipski50, 2024-12-17:
Will do that tomorrow. Can this be a LBT issue? Since I am located in Europe.

glipski50, 2024-12-17:
Should I update my X20R with the firmware TD-ISRMPro_3.0.1_20241209.frsk that has been published on GitHub?

SmartTommy, 2024-12-18:
> Should I update my X20R with the firmware TD-ISRMPro_3.0.1_20241209.frsk that has been published on GitHub?

it should does't matter. have you test the new TWR10 firmware?

glipski50, 2024-12-18:
I started testing the new firmware right now. I will send the result later today.

glipski50, 2024-12-18:
My first testrun had again a glich down to 0 V

[image]

Here's the logfile

[LiPo-Voltage-2024-12-18-09-52-29.zip](https://github.com/user-attachments/files/18180547/LiPo-Voltage-2024-12-18-09-52-29.zip)

I will do another testrun this afternoon my time

glipski50, 2024-12-18:
My second testrun today using a 4 cell lipo, a bit longer than 3 hours, with the test firmware is without any glich.

Tomorrow I will do another testrun using a 6 cell lipo.

SmartTommy, 2024-12-19:
> My first testrun had again a glich down to 0 V
> 
> ![screenshot-2024-12-18-42956](https://private-user-images.githubusercontent.com/116186946/396891469-113adb5a-b82b-4020-a27f-972b7bec2a51.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1Njc5MjcsIm5iZiI6MTczNDU2NzYyNywicGF0aCI6Ii8xMTYxODY5NDYvMzk2ODkxNDY5LTExM2FkYjVhLWI4MmItNDAyMC1hMjdmLTk3MmI3YmVjMmE1MS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxOVQwMDIwMjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZDRkYTY1MzQ4Nzc3NzM3YjVlNjYwYmJiOGMxMmNmZjA1ZmRkZmM5YjAzZjM3MTlhNTJiMGM0ODhmYmY4MjFiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.rCvcoh4LqfxmkbTKgYP3-045PSEfgXPf1-xflsHAt0Y)
> 
> ![image](https://private-user-images.githubusercontent.com/116186946/396889311-363875f2-6e8d-4db6-9624-c91bf84e7b4b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1Njc5MjcsIm5iZiI6MTczNDU2NzYyNywicGF0aCI6Ii8xMTYxODY5NDYvMzk2ODg5MzExLTM2Mzg3NWYyLTZlOGQtNGRiNi05NjI0LWM5MWJmODRlN2I0Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxOVQwMDIwMjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MGRmMjVhZDM5NDkwMTJhMDg5OGRiYTdlNjcwM2I0MTZlM2U4Mjc5NGNjMzE1YWFmZWU1ODY3N2NhODkzMmQ2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ieqcnPKgtL8pk2otLYPdTUEYaxT5vvsCT0yEWluvE6A)
> 
> Here's the logfile
> 
> [LiPo-Voltage-2024-12-18-09-52-29.zip](https://github.com/user-attachments/files/18180547/LiPo-Voltage-2024-12-18-09-52-29.zip)
> 
> I will do another testrun this afternoon my time

test with the "TWR10_3.0.1_FLVSS_TEST" firmware?

glipski50, 2024-12-19:
> test with the "TWR10_3.0.1_FLVSS_TEST" firmware?

Yes, both testruns yesterday have been with „TWR10_3.0.1_FLVSS_Test“ firmware and the testruns today will be as well with the test firmware you‘ve posted.

SmartTommy, 2024-12-19:
> Yes, both testruns yesterday have been with „TWR10_3.0.1_FLVSS_Test“ firmware and the testruns today will be as well with the test firmware you‘ve posted.

i have test this 6 hours, no glich. I will continue testing tonight to see how things will go tomorrow

glipski50, 2024-12-19:
Here again my configuration, just to be sure
X20R Ethos 1.6.0 RC10 LBT EU version
TW R10 with firmware TWR10_R3.0.1_FLVSS_Test you've posted
FLVSS

I have done testrun (Lipo 3S) today and I got glitches of cell 1 and cell 2 simultaniously down two zero, This time the logfile contained data in the fields the glitch occured of 0V and it looks the glitches of Lipo Cell 1 and Lipo Cell 2 have 
been longer.

[image]

[image]

Here's the log file 
[LiPo-Voltage-2024-12-19-12-57-16.zip](https://github.com/user-attachments/files/18199066/LiPo-Voltage-2024-12-19-12-57-16.zip)

Don't know why I get thoss glitches and you don't. Could it be the sensor FLVSS itselve? Anything else I could try other that reflashing ISRM-Pro, TW R10 and FLVSS?

I reflashed the TW R10 with the test firmware and started another test

SmartTommy, 2024-12-20:
> Here again my configuration, just to be sure X20R Ethos 1.6.0 RC10 LBT EU version TW R10 with firmware TWR10_R3.0.1_FLVSS_Test you've posted FLVSS
> 
> I have done testrun (Lipo 3S) today and I got glitches of cell 1 and cell 2 simultaniously down two zero, This time the logfile contained data in the fields the glitch occured of 0V and it looks the glitches of Lipo Cell 1 and Lipo Cell 2 have been longer.
> 
> ![screenshot-2024-12-19-53572](https://private-user-images.githubusercontent.com/116186946/397409549-c88a4df2-f0f6-4ee1-9059-d80dd7713afb.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NTM3NDQsIm5iZiI6MTczNDY1MzQ0NCwicGF0aCI6Ii8xMTYxODY5NDYvMzk3NDA5NTQ5LWM4OGE0ZGYyLWYwZjYtNGVlMS05MDU5LWQ4MGRkNzcxM2FmYi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMFQwMDEwNDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYTM4YTkzNzViZmZjZjliNGJiYTNhZjBhYzU5Y2ZjNWM1Y2Y4ODg2MjU2MTQwMWZjYTYzMjhkMTYwYjc4Mjk3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.s396YbbqDlkppJMOcSZV7niOuCkoiCIXP8bRhO4_Lfk)
> 
> ![Screenshot 2024-12-19 150138](https://private-user-images.githubusercontent.com/116186946/397409605-d31ea5b5-75b6-47dd-9b4a-deefa6e0d938.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NTM3NDQsIm5iZiI6MTczNDY1MzQ0NCwicGF0aCI6Ii8xMTYxODY5NDYvMzk3NDA5NjA1LWQzMWVhNWI1LTc1YjYtNDdkZC05YjRhLWRlZWZhNmUwZDkzOC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMFQwMDEwNDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MzU2NTVhNjMxNmZiNGY0NThmMzFjZjYxNzM1ZjkzNzhhNzM4OTY2ODJjNzg1OWUzNThiY2MxNzkzN2JjYTBlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sX0B2QLKKfVS1faFhUvysHbj1PZhMMbDpVT0C4C89o8)
> 
> Here's the log file [LiPo-Voltage-2024-12-19-12-57-16.zip](https://github.com/user-attachments/files/18199066/LiPo-Voltage-2024-12-19-12-57-16.zip)
> 
> Don't know why I get thoss glitches and you don't. Could it be the sensor FLVSS itselve? Anything else I could try other that reflashing ISRM-Pro, TW R10 and FLVSS?
> 
> I reflashed the TW R10 with the test firmware and started another test

After a night of testing, I get the glich and will identify and fix it as soon as possible

SmartTommy, 2024-12-24:
could you help me test the new firmware?
[TD-ISRMPro_3.0.1_FLVSS_X.zip](https://github.com/user-attachments/files/18237977/TD-ISRMPro_3.0.1_FLVSS_X.zip)

glipski50, 2024-12-24:
Will do that today and should be able to send the result during tomorrow my time.

SmartTommy, 2024-12-25:
> Will do that today and should be able to send the result during tomorrow my time.
> 
> So I update the ISRMpro module with the testfirmware und also have the TW R10 flashed with the testfirmwrae you've posted some days ago, correct?

yes

glipski50, 2024-12-25:
Yesterday night I did three testruns three hours each. I did not get any glitch.

I will do more testruns during the day today.

glipski50, 2024-12-25:
I did three test runs today each about four hours. This time I got two glitches again on VFR 2.4 GHz. The empty cell represents the first glitch. All other values of the log file are ok.

[image]

[LiPo-Voltage-2024-12-25-08-29-55.csv](https://github.com/user-attachments/files/18247244/LiPo-Voltage-2024-12-25-08-29-55.csv)

SmartTommy, 2024-12-27:
> I did three test runs today each about four hours. On one of the test runs I got two glitches again on VFR 2.4 GHz. The empty cell represents the first glitch. All other values of the log file are ok. The other test runs have been without any glitch.
> 
> ![Bildschirmfoto 2024-12-25 um 22 16 38](https://private-user-images.githubusercontent.com/116186946/398600653-b953accc-ff9a-433f-95e2-230cca3ccd8e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyNjk5MzEsIm5iZiI6MTczNTI2OTYzMSwicGF0aCI6Ii8xMTYxODY5NDYvMzk4NjAwNjUzLWI5NTNhY2NjLWZmOWEtNDMzZi05NWUyLTIzMGNjYTNjY2Q4ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyN1QwMzIwMzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMzJmZWY4MTk5Mjc4NDBmOWI3NWU5ZjNkYmVkMTMyN2Y5NGJjYTZkM2NlZDYzNGY2YjQxN2QxODc5NTM1MGY5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Zign1qn4FbDHnRps7Aty2VcNH9TLmPT5lGbciwXyD-8)
> 
> [LiPo-Voltage-2024-12-25-08-29-55.csv](https://github.com/user-attachments/files/18247244/LiPo-Voltage-2024-12-25-08-29-55.csv)
> 
> I won't be able to do any tests for the next four weeks.

Thank you for your test, and i think i found the reason report the error FLVSS value. and  start to fix the empty value issue, i will post it to you when fixed that.