FrSkyRC/RF_Feedback_Group #75: RB 25S no access to ADV sensors via S.Port and FBUS

3wintro, 2024-12-22:
RB 25S via S.Port and FBUS no access to ADV sensors since update to FW 3.0.1 TD-ISRM on XES. No matter which TD series receiver. Setting the ID is no longer possible. It only works when plugged directly into the receiver.

SmartTommy, 2024-12-23:
> RB 25S via S.Port and FBUS no access to ADV sensors since update to FW 3.0.1 TD-ISRM on XES. No matter which TD series receiver. Setting the ID is no longer possible. It only works when plugged directly into the receiver.

did it works before you update the FW 3.0.1?

3wintro, 2024-12-23:
I just tried it again, all flashed to the old status. Does not work either.

SmartTommy, 2024-12-23:
> I just tried it again, all flashed to the old status. Does not work either.

https://github.com/FrSkyRC/RF_Feedback_Group/issues/66#event-15544226257

closed 2024-12-23 by SmartTommy