FrSkyRC/RF_Feedback_Group #76: Low fsk rssi readings since 1.6.0/3.0.1

Joshadam0057, 2024-12-23:
Was getting critical rssi warnings as close as top of laumch (dlg) reverted back to 1.5.19 and all associated rssi remained in the 60-70s at its lowest8

SmartTommy, 2024-12-23:
> Was getting critical rssi warnings (tw gr6) as close as top of laumch (dlg) reverted back to 1.5.19 and all associated ...rssi remained in the 60-70s at its lowest8

hi, what's your mean "as close as top of laumch"?

Joshadam0057, 2024-12-23:
Im sorry.. meant to elaborate on that...i was flying dlg so "top of launch"
= apex of throw....50 meters or so, straight up

SmartTommy, 2024-12-23:
> meant to elaborate on that...i was flying dlg so "top of launch" = apex of throw....50 meters or so, straight up

did you mean you get RSSI waring when your DLG 50m distance to you? but when you back to 1.x.x rf firmware, it works normal?

Joshadam0057, 2024-12-23:
Correct

SmartTommy, 2024-12-23:
could you send us your log file? i want to know "rssi warnings" coming from whitch band, FSK or LORA or both two band.

> Correct

 could you send us your log file? i want to know "rssi warnings" coming from whitch band, FSK or LORA or both two band.

SmartTommy, 2024-12-23:
> Correct. I even tried to pull the antenna out.of the sides of nosecone (whiskers) it didnt really help

could you send us the log file?

Joshadam0057, 2024-12-23:
> Unfortunately, the logs didnt save, maybe because it was full..ive also
> since cleared the logs...it was definately the fsk band though

The.logs.may actyally be there

clintc2000, 2025-01-03:
I had similar experience after update to 1.6.0/3.0.1 TWGR6.  RSSI warnings/critical/telem lost at 2-300m in air and about 50m on ground for the fsk signal.  lora stayed in the 80+.  No loss of control.  I'll try to get some logs next time I'm out flying

Joshadam0057, 2025-01-05:
Feel a little better now that im.not the only one

ilioSS53, 2025-01-17:
Hello,
Here the same.
X20S and AP GR6 in easystar 3
Update to 1.6.0 and 3.0 direct after handlaunch low rssi warning?
Immidiate landing as precaution. Danger when flying with this warning.
1,6.0 still in RX  rf went back to previous version. Now 9k.
Looks like eu lbt is malfunction.
A video from Skybound RC jose most probably FCC config his system to 1.6 & 3.01 apperently no problem.
Is there a official statement from FrSky related to this malfunction?

ilioSS53, 2025-01-17:
Also mentioned here that internal and/or external antenne mistake coult be made.
For this purposses i install an infinity external antenne on 2.4 , always an antenne fitted.

SmartTommy, 2025-01-20:
> Also mentioned here that internal and/or external antenne mistake coult be made.
> For this purposses i install an infinity external antenne on 2.4 , always an antenne fitted

what's mean " Also mentioned here that internal and/or external antenne mistake coult be made"?

could you provide a detailed description?

ilioSS53, 2025-01-20:
Howe smart is SmartTommy?
Only replay with questions.
Never a awnser.

strgaltdel, 2025-01-22:
I’m also a user who is about to switch to 1.6.x & RF 3.x.

I've an IT job and must admit that , in your eyes, I’m probably just as "un-smart" as smartTommy.

At our company, issues are handled in such a way that the developer is responsible for fixing his software errors as quickly as possible and obtaining the necessary information if it’s not already available.
The more detailed and accurate the error descriptions and information, the more effective the resolution process.

The user (or the person who opens the issue in the ticket system) is required to describe the error as precisely, extensively, and in as much detail as possible.
Not all users provide the necessary information right away, so the developer asks follow-up questions.

In this issue, as the issue owner, I would have asked the same questions as smarttommy, since some things were only vaguely described, which can quickly lead to misunderstandings and hinder problem resolution.

That is also what I call professional work of my employees.

jm2c

KAID-YANG, 2025-01-23:
> Hello, Here the same. X20S and AP GR6 in easystar 3 Update to 1.6.0 and 3.0 direct after handlaunch low rssi warning? Immidiate landing as precaution. Danger when flying with this warning. 1,6.0 [...]

Could you provide a video of the issue to help us reproduce the problem? We tested X20 LBT 1.6.0/3.0.1 and AP GR6 3.0.1, and no low RSSI warnings were observed.

clintc2000, 2025-01-25:
[BAMF2-Green-2025-01-26-11-20-59.xlsx](https://github.com/user-attachments/files/18548335/BAMF2-Green-2025-01-26-11-20-59.xlsx) shows rssi drop first on twin pro module, then same rx from internal module.  distance ranged from 1-2 feet away to 8-10 ft away

KAID-YANG, 2025-01-26:
> [BAMF2-Green-2025-01-26-11-20-59.xlsx](https://github.com/user-attachments/files/18548335/BAMF2-Green-2025-01-26-11-20-59.xlsx) shows rssi drop first on twin pro module, then same rx from internal module. distance ranged from 1-2 feet away to 8-10 ft away. power was at 10mw on external module. But RSSI didn't seem different between internal or external

Is this the same issue with severe RSSI warnings at close distances of 10-20m when using the Twin X Lite transmitter combined with the Twin Lite Pro external module?

clintc2000, 2025-01-30:
> > [BAMF2-Green-2025-01-26-11-20-59.xlsx](https://github.com/user-attachments/files/18548335/BAMF2-Green-2025-01-26-11-20-59.xlsx) shows rssi drop first on twin pro module, then same rx from internal module. distance ranged from 1-2 feet away to 8-10 ft away. power was at 10mw on external module. But RSSI didn't seem different between internal or external
> 
> Is it the same issue of severe RSSI warnings at a close range of 10-20 meters with the combination of the Twin X Lite transmitter and the Twin Lite Pro external module?

yes

bsongis, 2025-03-19:
> Howe smart is SmartTommy? Only replay with questions. Never a awnser.

Very smart post! I hope you are not native english speaker or it is just a shame to write that! Your posts are plenty of typos / mistakes and they are very difficult to understand for French or Chinese people.