FrSkyRC/RF_Feedback_Group #78: TDSR12 No RX Bat Telemetry

airxplore, 2024-12-23:
FW 1.6
TD SR12 FW 3.01

After updating firmware in both TX & RX I have lost the RXBATT voltage sensor. Have re-installed firmware, done factory reset, discovered new sensors but still nothing.

Any suggestions.
Thanks

4qr1hu9n7, 2024-12-24:
Hello;
I have a TD-SR18 rx. Performed the mandatory update. Re registered rx okay. Rebound rx, okay. 

RESULTS;
No telemetry.
NO responses from most of the channels. And those that do respond do so incorrectly!
Why am I not surprised!!

SmartTommy, 2024-12-25:
> Hello; I have a TD-SR18 rx. Performed the mandatory update. Re registered rx okay. Rebound rx, okay.
> 
> RESULTS; No telemetry. NO responses from most of the channels. And those that do respond do so incorrectly! Why am I not surprised!!

which Radio you used? X20PRO? X20RS?  did you upgrade your radio to 1.6.0 and both your Tx and Rx to 3.0.1?

airxplore, 2024-12-26:
> [TDSR12_3.0.1_RC2.zip](https://github.com/user-attachments/files/18224639/TDSR12_3.0.1_RC2.zip)

Can confirm that this has sorted the issue, works great thanks and have noticed its now in the downloads section of Ethos suite. Great work guys

closed 2024-12-26 by airxplore

4qr1hu9n7, 2024-12-26:
That's great. I post a question, and the answer was confusing. Is it mandatory to update the TD-SR18 receivers in order to use the latest Ethos release, which is 1.6.0?

airxplore, 2024-12-26:
Yes generally I always keep everything updated, if not I generally find you can get some strange  outcomes

4qr1hu9n7, 2024-12-26:
The instructions say the receiver must have the six-axis gyro recalibrated in order to get the latest improvements for the TD-SR18.
My receivers are extremely difficult to remove from the model. Also. I have no interest in these improvements. Do I still need to update the TD-SR18 in order to use Ethos 1.6.0?

Richrfl, 2024-12-26:
> The instructions say the receiver must have the six-axis gyro recalibrated in order to get the latest improvements for the TD-SR18. My receivers are extremely difficult to remove from the model. [...]

I would suggest that if your balance between removing the RX from the model and Suite 1.6.0 weights on the inconvenience of removing the RX, and you do not need the enhancements, then roll back to a previous version of Ethos that you know it's compatible with your RX FW. That is really easy to do.

4qr1hu9n7, 2024-12-29:
Latest update. So, I installed 1.6.0. All of my receiver's function properly. Initially it looked like the TD-SR18 receivers were working. Upon further investigation, I noticed that the TD-SR18 had "NO" telemetry. This was true on my other TD-SR18 receivers 
I then decided that I have no other option but to remove the TD-SR18 receivers and perform the recommended setup. The firmware for the TD-SR18 installed without issue. The setup was easy and very quick. The TD-SR18 receivers now function as before the update. The difference is that they are now updated and work properly with full telemetry. The Ethos suite 1.6.0 is much easier to navigate when doing an initial receiver install. Glad I did it.