FrSkyRC/RF_Feedback_Group #79: x20rs s port

stubugs, 2024-12-23:
ugraded x20rs to 1.6   and   int module to 3.0.1 boot loader and system and luas etc
upgraded Tw r6 to 3.0.1
upgraded Tw sr8 to 3.0.1 performed reset
both working great with tx
but
Tw r6 picks up s port  150 amp current sensor all telemetry showing and lipo sensor 
plug same 150 s port sensor in Tw sr6 all telemetry of rx comes on but no 150 amp current sensor showing also tried in tw sr6 lipo sensor sport witch does not register either tried another lead on rx too still not working but red led does light up on 150 and lipo sensor

SmartTommy, 2024-12-24:
Hi, we have never produced "TWSR6" before,do you mean "TDSR6" or "TWGR6"?

stubugs, 2024-12-24:
The frsky twin sr8 reciever

SmartTommy, 2024-12-24:
> The frsky twin sr8 reciever https://www.frsky-rc.com/tw-sr8/ It worked on previous firmware

[image]
 i test the firmware 3.0.1 with the TWSR8 connect to FAS150, it seems work well

SmartTommy, 2024-12-24:
> The frsky twin sr8 reciever https://www.frsky-rc.com/tw-sr8/ It worked on previous firmware

[image]

i test the firmware 3.0.1 TWSR8, connect s.port to the FAS150, it works well

stubugs, 2024-12-24:
[image]
 These are what I use but do not work i have 4 of them

mawzthefinn, 2024-12-24:
You have the FAS150 connected to the 4-pin S.Port/FBus header on the TW SR8 and you have Telem. Mode set to S.Port in Receiver Options, right?

stubugs, 2024-12-24:
Yes and I tried another 4pin to servo lead same result

KAID-YANG, 2025-01-15:
Hello, could you please provide the video of the issue to help us investigate the problem?

stubugs, 2025-01-15:
I've found the problem  with the tw sr8 set to quick setting it's OK external sensors work but in the lua if you set quick and adv it looses all external sensors but keeps rx sensors

stubugs, 2025-01-23:
[image]
[image]
[image]
[image]
When enable  adv you loses telemetry  sensors lipo in this case

KAID-YANG, 2025-01-24:
This is normal. When you enable ADV config, the S.PORT on the TW SR8 is converted into an ADV configuration interface and no longer transmits telemetry data. You need to disable it, power off, and then power on again for it to work normally.

stubugs, 2025-01-24:
Ok thankyou

closed 2025-01-24 by stubugs