FrSkyRC/RF_Feedback_Group #80: Rx voltage

lotus1235, 2024-12-23:
Hello, I fly sailplanes and am wiring the wing servos. Flaps and ailerons in each wing. To reduce the number of wires, available space is limited, can I us the (+) and (-) to supply power to the the other servo, reduces the number of wires. I am concerned that I might not have enough voltage for both servos coming off the common 2 leads. Running xact 5700 with 8 volts to the gr8 rx.
Thanks

SmartTommy, 2024-12-24:
could you tell me how many servos you connect to your gr8 RX?  could you show us the diagrammatic sketch?

lotus1235, 2024-12-24:
6 servos, xact 5600, 8.0 volts from 3s lipoSent from my Verizon, Samsung Galaxy smartphone

SmartTommy, 2024-12-24:
> 6 servos, xact 5600, 8.0 volts from 3s lipo

did you mean you use 3S lipo---->BEC--->8.0V----> 6 5600(5700?) Servos and Rx?

what is your BEC type? is it stronger enough to power your servos?

lotus1235, 2024-12-24:
No bec, esc 60 ampSent from my Verizon, Samsung Galaxy smartphone

SmartTommy, 2024-12-24:
please make sure your power for the Rx and the Servos > 0.3A + 1.2A(Stall Current)* number Of servos

lotus1235, 2024-12-24:
Thank you Tommy.

lotus1235, 2024-12-24:
8 amp bec, 20 amp max for .5 seconds

closed 2024-12-25 by SmartTommy