FrSkyRC/RF_Feedback_Group #83: RB25S gyro

sigurdma, 2024-12-23:
Hello, anyone tried the RB25S with 1.6.0 and 3.01 TR pro RF - X20pro?
I have updated the RB25S to the latest xx511.frsk.
I cant connect with the lua script 3.0.0 for RB25S.

Nicholas-Luoyi, 2024-12-24:
If I'm not mistaken. RB25S should still use the script of version 2.1.1

sigurdma, 2024-12-24:
Its two choices ith seems, tried both under devicec 2.1.1 and the separate 3.0.1. Se pictures.
No contact on both.
[image]
[image]

Nicholas-Luoyi, 2024-12-25:
Well. Both are OK here. What is your receiver?

sigurdma, 2024-12-25:
I have used TD R10 on Fbus on pin 10 to RB25S RX1.
I then changed to RB30 (with gyro) and TD R10 and pin 10 to Sbus and pin 9 to smart port. So far no connection to gyro setup.
I plan to try R8pro rx later today on RB30.

sigurdma, 2024-12-25:
I tried now with an old ACCST RX8Rpro. Everything works ok with this rx. In both device manager and lua 3.0.1.
Only when i try RB30 in Device manager there is only the config menu where I ca turn on/off gyro and things. But there are no menu to adjust gain an reverse ++. It should be there?

sigurdma, 2024-12-25:
Some news here: I discovered that on Fbus from TDR10, I must use Rx2 or Rx3 on the RB25 (not Rx1 that has smart port only it seems). So it works ok now on Rx2..
Still can not adjust gyro on RB30, will make a separate isse.

closed 2024-12-25 by sigurdma