FrSkyRC/RF_Feedback_Group #86: X20S + TD-SR12 no Rx VFR sensor in telemetry

mszulen, 2025-01-02:
Both (tx and rx) are on the latest firmware (3.0.1 on TX and 3.0.1 RC2 on RX).
Latest Ethos 1.6 FLEX (using 868 EU band)

VFR 2.4G and VFR 900M are visible and no Rx VFR.
Reflashed both and re-discovered the sensors to be sure that all is ok.
No change. Still, no Rx VFR

Also, the RX sensor is displayed twice (for each module/band). Is that correct?

[image]
[image]

bsongis-frsky, 2025-01-02:
@SmartTommy I think it is for you?

rburrow87, 2025-01-02:
I also lost the RxBatt sensor with my TD R12 (non-stabilized) using 3.0.1 RC2 RF firmware on TX and RX. Using an X20RS with Ethos 1.6.0.

mszulen, 2025-01-02:
> I also lost the RxBatt sensor with my TD R12 (non-stabilized) using 3.0.1 RC2 RF firmware on TX and RX. Using an X20RS with Ethos 1.6.0.

Interesting - RC2 was what brought the RXBatt sensor back in my case. I only have problem (confirmed by others) with missing Rx VFR sensor.

rburrow87, 2025-01-02:
Oh my bad, I totally misread the title and message, lol.

Someone really needs to run all the firmwares on all the receivers at least once before publishing even as a beta firmware.

SmartTommy, 2025-01-03:
[TD-ISRM_3.0.1_RC2_RXVFR.zip](https://github.com/user-attachments/files/18295871/TD-ISRM_3.0.1_RC2_RXVFR.zip)

@mszulen ,please try this, i forget add the VFR to the TD-ISRM_RC2.

mszulen, 2025-01-03:
> [TD-ISRM_3.0.1_RC2_RXVFR.zip](https://github.com/user-attachments/files/18295871/TD-ISRM_3.0.1_RC2_RXVFR.zip)
> 
> @mszulen ,please try this, i forget add the VFR to the TD-ISRM_RC2.

This one works. I've got the RXVfr along with a bunch of other, new sensors (accelerometer data)
Thanks!
[image]

xet303, 2025-01-03:
With this new version of TD-ISRM 3.01_Vfr me too RX Vfr has just appeared.
Test on TD SR10 & TD MX
Thanks

closed 2025-01-09 by SmartTommy