FrSkyRC/RF_Feedback_Group #87: Servo Matching / Balancing with stabilsation

airxplore, 2025-01-02:
Really liking where FrSky are going with the stabilised receivers, the interface has come on massively from the original stab receivers. Great work!!!

For the larger models / jets is there going to be a way to do servo matching within stabilsation, as once the gyro starts deflecting the control surfaces it is no longer taking into account the servo matching one would of performed in "Balance Channels"

airxplore, 2025-01-02:
I feel that by allowing the RX Stab system to incorporate a servo balancing system would really attract wider and stronger audience. 

Don't get me wrong the system has improved massively recently, but because it's improving all the time, I thought I would input my 2 pence.

Keep up the great work guys

bsongis, 2025-01-02:
@SmartTommy will decide if it is possible on the RX, I will adapt Ethos. It won't be easy though, we need the function to setup the balance curve! I think it will have a low priority anyhow :)

SmartTommy, 2025-01-03:
it is far more complex, the current SRX MCU has no enough resources to implement. We will consider trying this in the next generation of products