FrSkyRC/RF_Feedback_Group #88: No Telemetry F-Port TD-MX

Beastyboy80, 2025-01-02:
Hello.
I am not receiving telemetry from my Speedybee FC via F-Port.
The problem currently only affects the TD-MX, an Archer RS ​​does not have the problem 

HF firmware is 3.0.1
but the old one also has the same problem
Stick signals arrive in the FC

SmartTommy, 2025-01-03:
what is your Speedybee FC type? F4 or F7?

KAID-YANG, 2025-01-23:
Hello, if you are using the Speedybee F4 flight controller and have connected the inverted S.Port interface, the Archer RS can work on the F4. However, the TDMX requires the flight controller to support inverted commands to function. Since the F4 does not support inversion unless there is an inverted signal port, it cannot be used.