FrSkyRC/RF_Feedback_Group #89: Bricked Archer Plus GR8 after update to 3.0.1

CodeWorksCZ, 2025-01-02:
Has anyone managed to brick **Archer Plus GR8** as I did? Yesterday, I decided to upgrade Tandem X20 to 1.6, which went well, however this happend to me while updating RX

Latest Ethos Suite - **OK**
Tandem X20 updated to 1.6 - **OK**
TD IRSM updated to the latest - **OK**

Then I tried to flash my **Archer Plus GR8 to the 3.0.1 (ArcherPlus_GR8_3.0.1.frsk) via OTA** and at the end of upgrade I got " **Update successful**" message.
Disconnected from TX, plugged in battery and now I only get  solid blue LED and nothing happens.

So I tried to update the FW via cable which came with the RX  and now I get **Device not responding** . 
When flashing (Flash ex. device), RX blue led comes on for few sec and then the error message appears

_**The only difference is that couple months ago I applied a recommended Patch (2 files) which appeared on Frsky website and is not there anymore**._

Now the RX does not response and not even boot up and all I get is blue LED light. I am almost 100% sure, that the FW I was downloding was correct for Archer Plus GR8 and not for SR8

**Anyone knows a way out of this or do I have a paper weight now?**

SmartTommy, 2025-01-03:
did you mean you can't flash the Rx by STK? maybe you need return the Rx to service center.

CodeWorksCZ, 2025-01-03:
No, I mean I can not flash the receiver from transmitter using  the cable that came with the RX

Alex00101, 2025-01-04:
This is another case outvof multiple reported. Unfortunately you do have a paperweight unless you send to the service centre.

I've done some testing and going to open an issue here soon. The only new angle your post is opening for me is that patch you mentioned. That could be one of the possible reasons.

CodeWorksCZ, 2025-01-04:
I suspect that the issue might be related to the patch because, as I mentioned, I am not aware of any mistakes on my part. Even if I had theoretically applied the wrong firmware, I should still have been able to fix it by flashing it again using the provided cable.

Unfortunately, I don't have an STK and will probably need to send it to a service center

Alex00101, 2025-01-04:
I tried STK (as well as from the transmitter) on mine and it doesn't help, if it is bricked that's basically it.
There is at least one user who has had the same issue with the receiver that didn't have the patch.

CodeWorksCZ, 2025-01-04:
Do you also get a solid blue light, when battery is connected as well as when trying to flash the firmware from TX?

Alex00101, 2025-01-05:
I'm not sure. As far as I recall, one had solid led and one no led at all.

Didier-L, 2025-01-05:
I had the same issue with a GR6+. Applied the recommended patch few months ago without any issue.
Updated to 3.0.1 via OTA and since then, rx not working at all : no led at all when supplied either by a battery (alone) or connected to tx.
I tried via Tx direct connection : rx not found
I tried also via STK with the same result : device not found.

I suppose that the bootloader is bricked... Is there any way (even by soldering wires on pcb) to flash it by another mean like STM32_flash software ?

CodeWorksCZ, 2025-01-05:
Yes, looks like the bootloader is bricked. I am not aware of any other way how to fix this but to send it to service center.

janekx, 2025-01-06:
Noticed FrSky about the issue...

yak55, 2025-01-09:
Hello, unfortunately I also have an SR8 plus with patch during the update to 3... lost.
OTA went through correctly - no error message.
Blue LED is now permanently lit.

Neither accessible with the STK adapter nor with the S port of the transmitter.
Lost the joy of OTA.

Greetings Eugen

KAID-YANG, 2025-01-23:
It is recommended to send the receiver to the nearest FrSky service center in the USA or Europe for repair.