FrSkyRC/RF_Feedback_Group #90: Update TD R18 to 3.0.1, lost the telemetry of receiver´s consumption

jasalca, 2025-01-03:
When upgrading to 3.0.1 the receiver TD R18 has lost the telemetry of receiver´s consumption.
All other telemetry data is correct

My radio X20 pro AW

SmartTommy, 2025-01-04:
what's your mean“receiver consumption"? the TDR10 only have RxBat1, RxBat2, ADC2, RxCurrent

jasalca, 2025-01-04:
Sorry, lost Rxcurrent.

SmartTommy, 2025-01-04:
> Sorry, lost Rxcurrent. My receiver TD R18

could you show us your TDR18 Hardware info?  (it will displayed in "info" page)

jasalca, 2025-01-04:
[image]

jasalca, 2025-01-04:
I updated the receiver via OTA. I re-registered it, paired it, I did a factory reset, and discovered the telemetry sensors.

SmartTommy, 2025-01-06:
> I updated the receiver via OTA. I re-registered it, paired it, I did a factory reset, and discovered the telemetry sensors.

Hi @jasalca , do you mean the "RxCurrent" is "0", and never changed? rather than no displayed?

SmartTommy, 2025-01-06:
[TDR18_3.0.1_RC2.zip](https://github.com/user-attachments/files/18312909/TDR18_3.0.1_RC2.zip)

Hi @jasalca ,please test this new firmware

jasalca, 2025-01-06:
> > Actualicé el receptor a través de OTA. Lo volví a registrar, lo emparejé, hice un restablecimiento de fábrica y descubrí los sensores de telemetría.
> 
> Hola@jasalca, ¿quieres decir que el "RxCurrent" es "0", y nunca cambió? en lugar de no mostrar?

Hello, no, when discovering sensors RxCurrent does not appear.

jasalca, 2025-01-06:
> [TDR18_3.0.1_RC2.zip](https://github.com/user-attachments/files/18312909/TDR18_3.0.1_RC2.zip)
> 
> Hola@jasalca, por favor, prueba este nuevo firmware

Then when I get home I will try it and tell you. Thanks a lot.

jasalca, 2025-01-06:
I have tried the new firmware TDR18 3.0.1RC2 and it is still the same. There have been no changes.
When I look at hardware info, 3.0.1 appears.
Is it normal for RC2 not appear?

SmartTommy, 2025-01-07:
> I have tried the new firmware TDR18 3.0.1RC2 and it is still the same. There have been no changes. When I look at hardware info, 3.0.1 appears. Is it normal for RC2 not appear?

yes, RC2 not displayed. could you flash your Rx back to the previously stable working version? This can help us determine whether it is a software problem or a hardware malfunction

jasalca, 2025-01-07:
Ethos 1.5.19  ,Modul TD-ISRM Pro 1.1.13 RX TDR18 1.0.12 .Everythig working properly!
[image]
[image]

SmartTommy, 2025-01-08:
Could you try flash the Tx & Rx to 3.0.1, and the ethos use current version(1.5.19)?

jasalca, 2025-01-08:
I flashed the TX and Rx to 3.0.1 Rc2 and I use 1.5.19 and everything working properly.
Then I flashed the Tx to 1.6.0 and now everything working properly too.
I don´t understand anything, but now all is ok!!

jasalca, 2025-01-08:
Thank you very much for helping me. The truth is that you do an exceptional job!!! Congratulations!!!

closed 2025-01-09 by SmartTommy