FrSkyRC/RF_Feedback_Group #91: Bug in Ethos Suite 1.6.0

BJR981S, 2025-01-03:
Bug in Ethos Suite 1.6.0

Ethos Suite 1.6.0 ignores the System General storage location settings. It does this for the Models folder and the Scripts folder.
It goes direct to the Radio Storage location and bypasses the configuration settings. This means that LUA manager and Model manager do not work correctly. I also believe that it does not convert your models to 1.6.0 format if your models are on the SD card. For some radios I have not tested yet (In Process) I believe it just corrupts the models. This is the Case for the X20Pro AW, X20Pro and the X20RS. This has been tested with these three models types with Multiple radios of each.

T16-Pilot, 2025-01-03:
@BJR981S 

You might want to post this in the ETHOS-Feedback-Community, this is the RF section ;)

By the way, this problem might be why the ExpressLRS Lua is not working in 1.6.0 on my X14...

Regards 
Simon

BJR981S, 2025-01-03:
> @BJR981S
> 
> You might want to post this in the ETHOS-Feedback-Community, this is the RF section ;)
> 
> By the way, this problem might be why the ExpressLRS Lua is not working in 1.6.0 on my X14...
> 
> Regards Simon

Thanks, I keep posting in the wrong forums doh.....I will shut this one down.

closed 2025-01-03 by BJR981S