FrSkyRC/RF_Feedback_Group #92: Running 1.6.0 on X20S and unable to bind a S8R Rx and unable to configure SR8+

frskyorbust, 2025-01-04:
Update X20S to 1.6.0 with no issues. Had a new SR8+ and updated the Firmware to latest as well as the lua scripts on the TX. I was able to bind the RX as ACCST not ACCESS also I DO NOT have the rest option on the script. I am able to access the StabConfig script or the RX with no issue. SO the RX is bound and able to set the channels and throws BUT no able to set/reverse the stabilization throws. Please let me know the proper scripts and the proper locations. I have followed everything I found and just not sure. Currenty using a X8R which is rock solid but no stabilization.

With all of this said. I have had NO issues configuring a SR8+ or a S8R before 1.6.0

SmartTommy, 2025-01-06:
> Update X20S to 1.6.0 with no issues. Had a new SR8+ and updated the Firmware to latest as well as the lua scripts on the TX. I was able to bind the RX as ACCST not ACCESS also I DO NOT have the rest [...]

I didn't understand what you meant. Can you make a video for us?
It seems that everything is working properly during our testing

LiangAndrea, 2025-01-06:
Uploading VID_20250106_111849_20251613208.mp4…

LiangAndrea, 2025-01-06:
https://github.com/user-attachments/assets/4e1234f1-7e1a-471f-b072-6dcbcb9e42fa

KAID-YANG, 2025-01-22:
The corresponding LUA scripts for the receivers are as follows:  
S8R: Device Config → SxR  
Archer Plus SR8+: Device Config → Stabilizer Config  

For Archer Plus SR8+, it can only bind using the D16 protocol. It is recommended to reflash the receiver firmware.