FrSkyRC/RF_Feedback_Group #93: lost two planes since 3.0.1

Jazzdoittt, 2025-01-06:
rf lost in 100m range, think of broken rf module, spend 80€ buying new one, downgrade to 1.5 version and works aparently ok...not happy with frsky

Jazzdoittt, 2025-01-06:
yes, in 20 meters no signal

WhiskeyGolf, 2025-01-06:
Also with different Reciver/Protocol ?

SmartTommy, 2025-01-07:
> rf lost in 100m range, think of broken rf module, spend 80€ buying new one, downgrade to 1.5 version and works aparently ok...not happy with frsky

what is the Radio and the Rx you used?

Jazzdoittt, 2025-01-07:
x18, tested with r10 pro, r10+ yand r9sx

SmartTommy, 2025-01-07:
> x18, tested with r10 pro, r10+ and r9sx

we changed the Code TD/TW, no update the R10Pro and R9SX, ACESS only changed the S function. We will test according to the situation you described

Jazzdoittt, 2025-01-07:
it's posible i have rf module malfunction? i mean can this module work but with very low power? that was what i think first so i buyed a new one (waiting to receive it), it seems to work better on 1.5 ethos but still fluctuate...

sigurdma, 2025-01-07:
Its possible with low power if you choose the external antennas output without physical antennas connected, and not the internal antennas in the RF menu. Then U could also destroy the rf output amplifier. I have put two 50 Ohm dummies on the output connectors for that reason.

KAID-YANG, 2025-01-09:
Hello, please provide the information from the "Info" screen: the receiver firmware version, ETHOS system version, and internal RF firmware version. Also, check if the internal RF antenna has been switched to an external antenna. If it is in external antenna mode, please install the corresponding antenna as instructed and test again to see if there are any issues.