FrSkyRC/RF_Feedback_Group #94: Loss Of Telemetry Sensors

CaptKremen, 2025-01-06:
After a (otherwise) successful update to V1.6, 3.0.1 etc. have lost LiPo and Current Sensors using  first one TD SR10 Rx then tried another TD SR10. This using a X20S and then a X20RS in turn. Tried Device Config to reset ID and AppID, no luck. Re-installed Rx Firmware via cable no change. All other sensors 'discovered' and transmitting data OK. Stabilisation and other features all working OK too. Any steers what to do? How to get LiPo & Current Sensors to be discovered. Checked all cables including polarity AND these and other similar LiPo and Current sensors work perfectly with all other FrSky type of Rx e.g. TD SR6, SR6, S6R, TW. Sensor firmware also checked and are current version. Puzzled ??

SmartTommy, 2025-01-07:
> After a (otherwise) successful update to V1.6, 3.0.1 etc. have lost LiPo and Current Sensors using first one TD SR10 Rx then tried another TD SR10. This using a X20S and then a X20RS in turn. Tried [...]

it works here. could you try back to 1.5.x and Tx/Rx back to the old version to seem if works?

KAID-YANG, 2025-01-22:
We tried to reproduce the issue on our side but failed. We would like to know which devices you have connected and how they are connected to help us reproduce the problem.