FrSkyRC/RF_Feedback_Group #95: Receivers "bricked" via OTA: summary

Alex00101, 2025-01-07:
This is a summary of multiple instances where users have their receivers rendered inoperable (aka “bricked”) when upgrading these receivers OTA. This has been reported by the non-insignificant number of users across different social media channels (including this GutHub). It is my understanding that some of these channels are attended by members associated with FrSky, so my assertion that this can not be attributed to isolated incidents can be easily verified.  

**Symptoms:**

1. Users update their receiver firmware to the latest version from FrSky. It was verified in most if not all cases that FrSky OTA upgrade instructions were followed, and correct firmware selected.
2. OTA update starts normally, and the update progress indicates successful OTA upgrade competition.
3. After the upgrade however, the receiver cannot be returned to an operational state: it does not allow users to register, bind or establish and RF link in any other way. Most of the receivers in that state also have solid blue led on (and nothing else). Others have not reported led status.
4. Attempts to restore the operational state of the receiver using information and tools available to users – by attempts to reflash the firmware using wired connection were unsuccessful. NOTE: There were other cases where wired reflash worked - this submission is not expected to cover that, it is purely focused on the issue when a user cannot recover the receiver (receiver “bricked”).
5. In some cases, reported response from the FrSky was to return the receiver to them for “test and repair”.
+

**Additional information:**

1. Looks like mostly the Archer Plus, TW and TD receivers are affected.  This may be due to the fact that with the latest RF firmware update (3.0.1) users started to update these receivers in bulk and only these three families received an update. I couldn’t find any reports of older Arche receivers being bricked.
2. Some receivers had FrSky “patch” applied to them (the one that was published few months ago than pulled back), some had not.
3. Attempts to recover the receiver by re-flashing the firmware via wired connection included a) From the radio (with receiver connected via FrSky cable), using “flash internal device”, b) from the PC, connected via STK tool, and c) from the Suite 1.6 , radio connected to the Suite,   receiver connected via FrSky cable, flash from the download manager. In all cases attempts were not successful, receiver was not responding and timed out/produced error at device reset stage. ****NOTE: There was one report when method c) was successful while method a) failed, however this could not be reproduced/tested as a valid recovery method****.

**Testing:**
I have set up my own testing (and sacrificed on of the receivers). The test involved two receivers (one older Archer R8 Pro and Archer Plus receiver). The test involved continuous OTA firmware update (2.1.14 for Archer and 1.0.11 to 3.0.1 for Archer Plus). On the 5th cycle Archer Plus got bricked with the symptoms described above and cannot be recovered. 
My conclusions:

- Issue can be easily reproduced.
- Hardware issue with the receiver – one of the provided suggestions – is unlikely.  If indeed it is the root cause, in my opinion it poses a greater issue than unreliable OTA because I assume a hardware defect can also manifest itself under the normal operational condition (flying).
- If I could reproduce the issue without having access to FrSky inventory of receivers and equipment, surely it can also be reproduced by FrSky.

May I please respectfully ask FrSky team to:

1. Confirm and recognize these symptoms as an issue not isolated incident. In my opinion the number of reports justify that.
2. Review the currently suggested method of dealing with the issue by sending the receivers back – given a) costs incurred by users – some of users do not have a local authorized dealer, b) inconvenience/uncertainty of turnaround time – some of users to not have an abundance of spare receivers, c) overall risks and perceptions about OTA as a whole.
3. Perform root cause analysis and advice users whether the root cause is identified, can be fixed, which firmware version does / will have the fix.
4. Advice users about the risks of performing OTA (assuming the issue is confirmed) until the issue is not fixed.
5. If possible, develop a method to recover bricked receivers by end users, if not for the current version, then for the future.

Alex00101, 2025-01-08:
Thanks for looking into this.
I have 2 receiver models bricked with OTA.
1st was Archer Plus SR8 with the patch installed, being upgraded to the last firmware version before 3.0.1 - i.e. I was flashing 1.0.11.
This was on X20AWPro with Ethos 1.5.x (sorry do not recall exact version) and ISRM 1.1.13.

2nd was Archer Plus 6H, flashing from 1.0.11 (latest before 3.0.1) to 3.0.1 with X20ProAW on Ethos 1.6 and ISRM 3.0.1. I believe this one also had a patch.

3rd was - as a part of a test described above - another Archer Plus SR8 on 1.0.11 being flashed with 3.0.1, Ethos 1.6, ISRM 3.0.1. I did not install the patch on it.

Please note other users reported other models being bricked such as GR family but I have not had my own experience with these.

HTH

KAID-YANG, 2025-01-08:
> Thanks for looking into this. I have 2 receiver models bricked with OTA. 1st was Archer Plus SR8 with the patch installed, being upgraded to the last firmware version before 3.0.1 - i.e. I was flashing 1.0.11. This was on X20AWPro with Ethos 1.5.x (sorry do not recall exact version) and ISRM 1.1.13.
> 
> 2nd was Archer Plus 6H, flashing from 1.0.11 (latest before 3.0.1) to 3.0.1 with X20ProAW on Ethos 1.6 and ISRM 3.0.1. I believe this one also had a patch.
> 
> 3rd was - as a part of a test described above - another Archer Plus SR8 on 1.0.11 being flashed with 3.0.1, Ethos 1.6, ISRM 3.0.1. X20AWPro. I did not install the patch on it.
> 
> Please note other users reported other models being bricked such as GR family but I have not had my own experience with these.
> 
> HTH

Which receiver was used in the OTA bricking case, and which remote controller was used with it? I will conduct tests on the corresponding versions of the receiver and remote controller.

Alex00101, 2025-01-08:
All 3 receivers were bricked, one AWPro X20 radio in all cases

CodeWorksCZ, 2025-01-08:
In my case  #89 I was using Tandem X20 (1.6), Archer Plus GR8 with applied patches , ISRM updated to the latest version via OTA

SmartTommy, 2025-01-10:
> For my case, I was using X20 Pro n°269 and updated via OTA a Archer Plus GR6 with applied patches. Note: the GR6+ worked perfectly after the application of the patches. Problem came when updating to 3.0.1
> 
> X20 PRO : firmware 1.6.0 EU Internal module : TD-ISRM Pro, HW = 1.0.3 ; FW = 3.0.1
> 
> No LED working after the update.

have you try to flash the Rx back to 1.x.x?

Alex00101, 2025-01-10:
@SmartTommy the whole point of this issue is that after these OTA updates with described symptoms, there is no way of flashing any version of firmware back even via cable, from the radio, from the STK tool, nothing. The RX is dead.

Didier-L, 2025-01-13:
> > For my case, I was using X20 Pro n°269 and updated via OTA a Archer Plus GR6 with applied patches. Note: the GR6+ worked perfectly after the application of the patches. Problem came when updating to 3.0.1
> > X20 PRO : firmware 1.6.0 EU Internal module : TD-ISRM Pro, HW = 1.0.3 ; FW = 3.0.1
> > No LED working after the update.
> 
> have you try to flash the Rx back to 1.x.x with the STK or "FLASH external device"?

Yes, I have tried everything possible. the rx is not found by either the tx (s-port) or the STK tool.

KAID-YANG, 2025-01-22:
It is recommended to send the receiver to the nearest FrSky service center in the USA or Europe for repair.

T16-Pilot, 2025-01-26:
Hi, what is being done about the "bricked receivers" issues?

I am waiting for an update from Frsky before i dare to update my X20 Pro.

I lost a turbine jet after updating my X20 Pro from 1.5.6 to 1.5.12 and the module from 1.1.12 to 1.1.13. Receiver was Archer SR10 Pro and R9 MX as a backup. Both with latest firmware.

Model file: 

[mini avanti.bin.zip](https://github.com/user-attachments/files/18551309/mini.avanti.bin.zip)

There was nothing in the update documentation that said that we need to register and bind the rx after the update. All servos moved as normal after the update.

The result for me was a flame out and crash.

After the crash i started looking at my log files, and there are some interesting things going on in the flights after updating.

I got the log files before and after the update to prove it.

There fore i am waiting for what has/is being done about these bricked receivers before i update...?

[image]
[image]
[MiniAvanti_Logs.zip](https://github.com/user-attachments/files/18551300/MiniAvanti_Logs.zip)

SmartTommy, 2025-02-05:
> Hi, what is being done about the "bricked receivers" issues?
> 
> I am waiting for an update from Frsky before i dare to update my X20 Pro.
> 
> I lost a turbine jet after updating my X20 Pro from 1.5.6 to 1.5.12 and the module from 1.1.12 to 1.1.13. Receiver was Archer SR10 Pro and R9 MX as a backup. Both with latest firmware.
> 
> Model file:
> 
> [mini avanti.bin.zip](https://github.com/user-attachments/files/18551309/mini.avanti.bin.zip)
> 
> There was nothing in the update documentation that said that we need to register and bind the rx after the update. All servos moved as normal after the update.
> 
> The result for me was a flame out and crash.
> 
> After the crash i started looking at my log files, and there are some interesting things going on in the flights after updating.
> 
> I got the log files before and after the update to prove it.
> 
> There fore i am waiting for what has/is being done about these bricked receivers before i update...?
> 
> Top picture is before update, bottom picture is after update. Screen shot of throttle channel.
> 
> [image] [image] [MiniAvanti_Logs.zip](https://github.com/user-attachments/files/18551300/MiniAvanti_Logs.zip)

We have conducted many OTA upgrade tests, but have not replicated this issue. received a receiver sent back by a user and found that the binary file of the bootloader been modified. We suspect that this issue may have been caused by excessive flashing. We have added address protection in the next version of the program to ensure that such issues do not occur again.

For receivers that have already experienced this issue, we recommend that users send them back to our repair center.

Alex00101, 2025-02-05:
Thanks for the update. As the OP may I please post some observations or questions. 
1. Observation. In all tests I performed,  the affected receivers were fairly new, 1 or 2 previous flashing before. I have my theory of why it happened but out if receivers to test.
3. Obsevation. I assume by "modified" you mean "corrupted during OTA" as for sure neither myself nor other affected users were messing with .frsk files.
4. Observation. Any protection is great idea including adress one.
5. Question. By program do you mean new firmware version that will be published? Or the new protection will be applied to only new receivers with factory flashed bootloader? 
Thanks
6

SmartTommy, 2025-02-05:
> Thanks for the update. As the OP may I please post some observations or questions.
> 
> 1. Observation. In all tests I performed,  the affected receivers were fairly new, 1 or 2 previous flashing before. I have my theory of why it happened but out if receivers to test.
> 2. Obsevation. I assume by "modified" you mean "corrupted during OTA" as for sure neither myself nor other affected users were messing with .frsk files.
> 3. Observation. Any protection is great idea including adress one.
> 4. Question. By program do you mean new firmware version that will be published? Or the new protection will be applied to only new receivers with factory flashed bootloader?
>    Thanks
>    6

For 4:  the next version(3.0.2) address verification will be added to ensure FLASH program be limited. published.

T16-Pilot, 2025-02-19:
@SmartTommy 

Is there an ETA for 3.0.2?

Latest for X20 Pro is from Dec 11. 2024

Best regards
Simon

SmartTommy, 2025-02-20:
> [@SmartTommy](https://github.com/SmartTommy)
> 
> Is there an ETA for 3.0.2?
> 
> Latest for X20 Pro is from Dec 11. 2024
> 
> Best regards Simon

I submitted 3.0.2 last month, and i will push to complete the testing ASAP

aalain8, 2025-04-15:
I broke an Archer GR8 receiver after an OTA update.
The OTA and cable update didn't work anymore.
v2.1.14 / v2.1.10 and v2.1.12 were the same.
One day I had the idea of downloading the first software version v2.1.0 and the cable update worked.
Then I tried the OTA update, which also worked.
It's safe to try...