FrSkyRC/RF_Feedback_Group #97: Neuron 8A SBEC no longer recognised after using 1.6.0 LUA

BJR981S, 2025-01-10:
Neuron 8A SBEC no longer recognised after using 1.6.0 LUA 

It functions but no Telemetry and can no longer be configured.

I was going to re flash the FW but FW is not available anywhere, no initial FW release can be found and there is no update available..

Where can I get the FW to attempt to fix?

bsongis-frsky, 2025-01-10:
@SmartTommy I think is for you? Feel free to reassign to us if needed!

SmartTommy, 2025-01-10:
> Neuron 8A SBEC no longer recognised

could you connect your SBEC .SPORT to your Radio and then use the "device config--->ESC--->SBEC" to read your BEC configuration?

BJR981S, 2025-01-10:
> > Neuron 8A SBEC no longer recognised
> 
> could you connect your SBEC .SPORT to your Radio and then use the "device config--->ESC--->SBEC" to read your BEC configuration?

Thanks for your support.

I have tried via an TWMX, and TW Mini. Have now tried with your suggestion. Pictures attached.

Still cannot be recognised . Also the Green light signifying that the S-Port is connected is not on.

[image]

 
[image]

[image]

BJR981S, 2025-01-10:
The FW file provided is not valid.

It says Read Error when trying to install.

SmartTommy, 2025-01-11:
> The FW file provided is not valid.
> 
> It says Read Error when trying to install.

could you try use the STK to upgrade?

BJR981S, 2025-01-11:
> could you try use the STK to upgrade?

Excellent. That was successful The FW loaded OK and the SBEC is working as normal again.

Was the FW provided by you the original release FW or was it a later update?

Now that I know I can recover the SBEC is it worthwhile testing the LUA 3.0.0 again?

I have another 5 units arriving next week for testing.

I was going to test the PhyID and AppID full option range to determine if there is another issue the same as with the default settings of the 8A SBEC with the RB25S latest test FW for S-Port fix.

The S-Port update for the RB35S worked fine with the RB35S default settings.

Thanks for your help and advice.

Didier-L, 2025-01-13:
To Frsky guys:
I have the 8A and 12A SBEC. Nice feature, they have both a programmable voltage and telemetry.
With telemetry, you get the current (ok) and the output voltage !!!!!!!
The output voltage is totally useless as it is equal to the one programmed and to the Rx Bat.
Then, it would be much more usefull to have the input voltage to get the battery status. Is it possible ?
For the time being, I am obliged to run a separate cable to A2 entry with is totally stupid taking into account the telemetry...
Thank you

BJR981S, 2025-01-13:
> To Frsky guys: I have the 8A and 12A SBEC. Nice feature, they have both a programmable voltage and telemetry. With telemetry, you get the current (ok) and the output voltage !!!!!!! The output [...]

I use a LiPo Sensor on the battery. Yes the input voltage will change on the SBEC as the battery depletes. but overall battery voltage is not as critical as individual cell voltage. And the output voltage does vary depending pn the load on the SBEC. Its not a true voltage source, at least that is my experience with them. You can use the Output voltage to tune the set voltage to get a more accurate output. My use of the BECs is in line with things like Retract controllers, Brake controllers,  normal servos and specialist servos that wont operate above 6.0 Volts they all have no alternate means of reporting their input voltage. I try an get the max voltage that the servos will operate at the RX to get best speed and torque.

But I agree that input voltage would be good in some implementations.

You should start a new enhancement Issue.

SmartTommy, 2025-01-14:
> To Frsky guys: I have the 8A and 12A SBEC. Nice feature, they have both a programmable voltage and telemetry. With telemetry, you get the current (ok) and the output voltage !!!!!!! The output [...]

Thank you for your suggestion, we will add the input voltage with the next firmware upgrade(only BEC14A, The BEC 8A hardware seems not support), if you want, i can send you a special test firmware

closed 2025-01-23 by SmartTommy

Didier-L, 2025-03-10:
> Thank you for your suggestion, we will add the input voltage with the next firmware upgrade(only BEC14A, The BEC 8A hardware seems not support), if you want, i can send you a special test firmware

Hello, I checked the firmware for SBEC 14A and I did not found any new fw including the voltage input telemetry.
I am ok to test a new special fw.
Thank you.

SmartTommy, 2025-03-11:
[SBEC_14A_1.0.2_Beta.zip](https://github.com/user-attachments/files/19179924/SBEC_14A_1.0.2_Beta.zip)

The new firmware is testing, and the ethos will add the input volt display next version.

Didier-L, 2025-03-11:
> [SBEC_14A_1.0.2_Beta.zip](https://github.com/user-attachments/files/19179924/SBEC_14A_1.0.2_Beta.zip)
> 
> The new firmware is testing, and the ethos will add the input volt display next version.

Thank you. I updated the fw with STK (not working via the tx). I still do not have the input voltage.

So, if I understand well, I will get it with Ethos 1.6.3 ? (I hope that I will not have to wait for Ethos 1.7 ;-) )