FrSkyRC/RF_Feedback_Group #98: Flashing tdr18 3.0.1 went wrong

rwijnhov, 2025-01-12:
I just flashed my rxses with 3.0.1. one newer tdr18 2 2.4 antennas flashed ok. The old one with one 2.4 didn't. All I get now is a red led no binding and no registering. Any ideas on how to fix this. Also a warning that flashing isn't always succesful as it appears not lets hope its not a paperweight. I see this warning an believe this happened to me https://forum.alofthobbies.com/index.php?threads/important-frsky-rf-version-3-0-1.4420/

rwijnhov, 2025-01-12:
flashing by cable gets it fixed
[image]

KAID-YANG, 2025-01-13:
Hello, please provide the information from the remote controller's info interface, including the system version of the remote controller and the version of the built-in RF module.

KAID-YANG, 2025-01-13:
And please provide photos of the unbindable TDR18 with its casing removed. Thank you!

closed 2025-01-23 by SmartTommy