FrSkyRC/RF_Feedback_Group #100: Rx TD R10 configurable channels ports

Jepelisa, 2025-01-14:
I have Tx XE-S Emitter in v1.0.6 and TD ISRM module in v3.0.1 and TD R10 receiver in V3.0.1.
After recording and bind ok (led green on and blue flashing)I want to change the receiver settings as indicated in the instruction manual. 
On my radio screen after tap on set of TDR1O I only have access to "option/bind/clear". I do not have "share" and "fligt data record"
If I go in "options" I see first "waitting for receiver" and then "no response from receiver"
Please help me!
[image]

mawzthefinn, 2025-01-14:
> The radio doesn't know which receiver options are available, because of the telemetry restriction. Not sure if it is easy to fix.

It should on current firmware, because the telemetry is forced onto the 2.4G link now for EU users in the case of 868M power over 25mW. The behaviour changed a couple versions ago.

SmartTommy, 2025-01-17:
The Rx Set data transmission via 900M band, telemetry  not be allowed when 900M Tx power higher than 25mW(ETSI  Standards). you can congfigure your Rx when Tx Power <= 25mW.

mawzthefinn, 2025-01-17:
> The Rx Set data transmission via 900M band, telemetry not be allowed when 900M Tx power higher than 25mW(ETSI Standards). you can congfigure your Rx when Tx Power <= 25mW.

Current TD firmware transmits telemetry on the 2.4G band when the 900M band is set with a power above 25mW for EU mode. So it should be possible to configure the receiver over the 2.4G link.

closed 2025-01-23 by SmartTommy