FrSkyRC/RF_Feedback_Group #101: Ethos 1.6.0 Sport sensors missing when powering model on

yak-54, 2025-01-15:
this has only started after updating 
i have tried redoing all the updates using the 2 other radios and flash tool update the receivers  many times

 its doing this on 2 of my plane that use TD Receivers X20 pro on 1.6.0 / isrm module is on 3.0.1 receivers  TD-r18 TD-sr18 and TD-r6  3 lipo sensors and Gas suite

some times  when powering on a model none of the lipo senors or the gas suite sensors show up 
even if power cycle 3 or 4 time still will not show

the only way i can get to come back on is
go into model  settings rf system TDR18 options and cycle the pins i have set for sport in on pins ( 17/18 ) 
i have noticed rssi is all over the place as  

here is a video

 https://youtu.be/XQa1AeUj-S4

[model-bin.zip](https://github.com/user-attachments/files/18415487/model-bin.zip)

SmartTommy, 2025-01-16:
> this has only started after updating i have tried redoing all the updates using the 2 other radios and flash tool update the receivers many times
>
> its doing this on 2 of my plane that use TD [...]

how many Rxs you used on one aircraft?

yak-54, 2025-01-16:
TD-R18 and TD-R6
the TD-R6 has telemetry disabled

yak-54, 2025-01-16:
and the other plane that has this issue as well it had TD-SR18 only

SmartTommy, 2025-01-16:
> TD-R18 and TD-R6 the TD-R6 has telemetry disabled

we test here, and the lipo dispalyed every times. could you try to remove the TDR6, only use the TDR18?

yak-54, 2025-01-16:
> > TD-R18 and TD-R6 the TD-R6 has telemetry disabled
> 
> we test here, and the lipo dispalyed every times. could you try to remove the TDR6, only use the TDR18?

did you test with gas suite as well ?

SmartTommy, 2025-01-16:
> > > TD-R18 and TD-R6 the TD-R6 has telemetry disabled
> > 
> > 
> > we test here, and the lipo dispalyed every times. could you try to remove the TDR6, only use the TDR18?
> 
> did you test with gas suite as well ?

yes
[image]
[image]
[image]

SmartTommy, 2025-01-16:
Gas suite connect to the Pin17 and FLVSS ADV connect to Pin 18

yak-54, 2025-01-16:
as soon as get home i will test it and remove TD-R6

yak-54, 2025-01-16:
ok removed the TD-r6 still did it some times
removed the gas suite still did it 
so i re flashed TD-R18 with a sport tool on my pc and re config it and fit it to the plane with the all the sensors 
has not done it on 20 power cycles   so re fitted  TD-R6  20 power cycles did not do it   
the only issue i getting now is sensors conflict

SmartTommy, 2025-01-17:
> ok removed the TD-r6 still did it some times removed the gas suite still did it so i re flashed TD-R18 with a sport tool on my pc and re config it and fit it to the plane with the all the sensors [...]

which sensors conflict?

yak-54, 2025-01-17:
FLVS ADV i use 3 off them

SmartTommy, 2025-01-17:
> FLVS ADV i use 3 off them
> 
> if i remove them the warring's stops

change your three FLVS ADV use different  PhyID and AppID, then you can slove this

yak-54, 2025-01-17:
all good now i have change them before i forgot check them after i update to 3.0.1

closed 2025-01-17 by yak-54