FrSkyRC/RF_Feedback_Group #102: X20rs

stubugs, 2025-01-15:
[image]
Using  frsky adv gps  time has to be offset by -4hrs to get correct time in uk

xet303, 2025-01-15:
I had noticed this phenomenon this summer to finally realize that when it also happened my home clock also synchronizing by GPS marked the same wrong time as my X20S. For a while the GPS position precision at these times seemed to me around 2 meters.
The phenomenon ended up becoming less and less frequent until disappearing. But I saw the phenomenon reappear 1 or 2 days ago.
Maybe we should transfer our problem to the American army... It's heating up on the planet right now!!??
I had made a post about it on RCG, I should find it again.

BJR981S, 2025-03-16:
> Maybe if sensor could show no of sats and hop information

Number of Sats is not all that meaningful. It depends on their position in the Sky. The Data that we need is HDOP and VDOP.

Horizontal and Vertical Degree of Precision.