FrSkyRC/RF_Feedback_Group #105: TW x Lite  RSSI low 1.6.0 and RX 3.0.1

jereitor, 2025-01-22:
Hi, I have a TX Twin X Lite S, I recently upgraded to Ethos 1.6.0, I also upgraded the RX TW SR8 (3.0.1) and ISMR (3.0.1). Since then my RSSI signal drops a lot even when I'm close, and I get low signal warnings, doing the test I can only get about 15 meters away. My club mate with the same TX and RX has the same problem and has crashed a plane. everything was working fine before the update.

SmartTommy, 2025-01-22:
> Hello I have had the same experience with an XE-S transmitter with a Twin Lite module installed. I updated to Ethos 1.6, updated the TD-ISRM to 3.0.1 updated the Lite module to 3.0.1, updated receivers to 3.0.1. All seemed ok until I went flying and immediately had critical rssi issues. Checked hardware at home and found I had no rf from one of the TW antennas. (Checked with IRC Immersion RF meter). I rolled back Ethos and the firmware to previous revision. It made no difference, I still had no rf from one side of the TW module. Has since been replaced under warranty and I have been holing off updating again until 1.6.1 is released. This morning I have emailed FrSky Service and asked them to check this out.
> 
> [image] [image]

could you flash the Tx back to 1.0.7 and check the Tx Power again?

kramgrand, 2025-01-22:
I no longer have the defective TW Lite module as it was returned to the supplier as faulty under warranty.   

But during the process of fault finding.  I rolled back Ethos to 1.5.9, The TD-ISRM of the transmitter to TD 2.2.7. and the **firmware of the TW Lite module back to 1.0.7**   It made no difference, I still had no rf from one side of the TW module.

I am now using a new TW Lite Pro module.

SmartTommy, 2025-01-22:
> I no longer have the defective TW Lite module as it was returned to the supplier as faulty under warranty.
> 
> But during the process of fault finding. I rolled back Ethos to 1.5.9, The TD-ISRM of the transmitter to TD 2.2.7. and the **firmware of the TW Lite module back to 1.0.7** It made no difference, I still had no rf from one side of the TW module.I am now using a new TW Lite Pro module. The picture below shows the output of the old module at -8db . There is no output.
> 
> [image]

does your new TW-Lite module TxPower normal?

drewsho, 2025-01-22:
> I no longer have the defective TW Lite module as it was returned to the supplier as faulty under warranty.
>
> But during the process of fault finding. I rolled back Ethos to 1.5.9, The TD-ISRM of the [...]

Hi kramgrand, that sounds a little bit like crashing Twin Lite Pro Module with Firmware 3.0.1. I'm having the same issue and after flashing back to Firmware 1.0.7 and Ethos 1.5.19 and also flashing back the receiver (don't know exactly the Firmware revision) nothing changes. I notice that the blue LED is on at the receiver. That means that LoRa connection is established. But not sure if this was also before the firmware 3.0.1

SmartTommy, 2025-01-22:
> I no longer have the defective TW Lite module as it was returned to the supplier as faulty under warranty.
>
> But during the process of fault finding. I rolled back Ethos to 1.5.9, The TD-ISRM of the [...]

does your new TTW Lite Pro module RF POWER normal now?

kramgrand, 2025-01-22:
YES.  I have rf output from both antennas. Checked with IRC Immersion RF meter.

SmartTommy, 2025-01-22:
> YES. I have rf output from both antennas. Checked with IRC Immersion RF meter.

1.your new TW-Lite Pro with the firmware 1.0.7 has normal rf power,
2. do you test your nnew TW-LITE Pro  rf power with the 3.0.1?

SmartTommy, 2025-01-22:
> NO.... If, as it looks like, 3.0.1 is breaking the TW Lite Pro, I do not think the uk supplier would be keen to supply me with another TW module under warranty. I will leave that testing to FrSky.

i test it here, it works normal. could you send your 3.0.1 firmware to me?

kramgrand, 2025-01-22:
This is the compressed file off my XE-S.   I tried to just send the .frsk file but it is not a permitted file type on Github. It was initially downloaded from the Ethos Suite download manager

[3.0.1 TW-Lite.zip](https://github.com/user-attachments/files/18503980/3.0.1.TW-Lite.zip)

SmartTommy, 2025-01-22:
> This is the compressed file off my XE-S. I tried to just send the .frsk file but it is not a permitted file type on Github. It was initially downloaded from the Ethos Suite download manager
> 
> [3.0.1 TW-Lite.zip](https://github.com/user-attachments/files/18503980/3.0.1.TW-Lite.zip)

[image]

[image]

flashed the the firmware you post,  it works as expect.

SmartTommy, 2025-01-22:
> Hi, I have a TX Twin X Lite S, I recently upgraded to Ethos 1.6.0, I also upgraded the RX TW SR8 (3.0.1) and ISMR (3.0.1). Since then my RSSI signal drops a lot even when I'm close, and I get low [...]

i will check it and reply. thank your for your feedback.

kramgrand, 2025-01-22:
If I run it again and it breaks it. What do I do then?

SmartTommy, 2025-01-22:
> If I run it again and it breaks it. What do I do then?

[TW-LITE_3.0.2_RC1.zip](https://github.com/user-attachments/files/18504300/TW-LITE_3.0.2_RC1.zip)

please try the special firmware for you.

kramgrand, 2025-01-22:
Done.  RF power looks okay.

[image]
[image]
[image]

SmartTommy, 2025-01-22:
> Done. RF power looks okay.
> 
> [image] [image] [image]

Thank. i will do more test tomorrow.  please feel free to let me know if  you have any new discoveries,

kramgrand, 2025-01-22:
I do not plan to do the other updates until Ethos 1.6.1  /  1.6.2 is released.   
I do hope others listed above with this problem are able to find a repair solution, as I do not think running the firmware on broken module fixes the problem.

drewsho, 2025-01-22:
I also tested this Firmware 3.0.2 at my Twin Pro Lite Modul. It changes nothing. RSSI decrease rapidly when the distance between TX and RX increases. Here are two pictures

[image]
[image]

jereitor, 2025-01-22:
> I also tested this Firmware 3.0.2 at my Twin Pro Lite Modul. It changes nothing. RSSI decrease rapidly when the distance between TX and RX increases. Here are two pictures
> 
> [image] [image]
> 
> 1 and a half Meter away from the RX RSSI shows only 76db RSSI

Exactly the same thing happens to me.

jereitor, 2025-01-22:
So far I have no answer from FRSKY support. Is there any place to download the beta versions of ISRM and Receiver ?

kramgrand, 2025-01-22:
I have heard back from FrSky support.

[Ethos 160 and  301 Firmware upgrade.pdf](https://github.com/user-attachments/files/18510484/Ethos.160.and.301.Firmware.upgrade.pdf)

drewsho, 2025-01-22:
Hi Kramgrand,

I'm also reading at the German FrSky forum. Next week I want to try to replicate your measurements on my Twin Lite Pro module. I don't have a measuring instrument like you, but we'll find out whether it's transmitting or not. If I have any new information, I'll write here. And yes, everything is closed in China during CNY.

kramgrand, 2025-01-25:
Today was the first time I was able to go flying since receiving the warranty replacement Twin Lite Pro module.  Everything is now working as expected with NO rssi warnings or range issues reported.

drewsho, 2025-01-25:
> Today was the first time I was able to go flying since receiving the warranty replacement Twin Lite Pro module. Everything is now working as expected with NO rssi warnings or range issues reported.

Hi MarK,

can you do me a favor and tell me which color the Status LED from RX is shown?

My Receiver shows Blue Color what means that LoRa is active.

clintc2000, 2025-01-25:
I have similar results.  Twin X Lite S.  Both internal and external module I see decreased RSSI for the fsk side while the lora side is fine.  drop from high 90s to mid 80s 10 feet away.  Should I try the 3.0.2 firmware?

clintc2000, 2025-01-25:
I was able to bind same rx on 3.0.1 fw to another twin x lite with twin pro module on 1.5.x and 1.0.7 and it seems to work no issues on the bench.  vfr and rssi are single sensors, so I assume they are the combined values (same as rx vfr on new fw?)   I was not able to reproduce the drop in rssi under same conditions it drops with 3.0.1.  I have context coming up in 3 week, so I don't want to update my second radio until there is more confidence this has been resolved

Jazzdoittt, 2025-01-28:
Same issues here. Firts i think i burned the rf module, changed it (85€), then buy twin little pro and twr8, and power is about 75db at 2meters, either internal and external,  900 band is fine, lora is fine, upgrade to 1.6.1rc3 and the 3.0.2 firm posted here, and same results... wtf, now i don't know if two antennas are working and go to fly with fear (i already crased and destroy a b25 plane)... always happy with frsky but now...  i'm using flex fyi.

kramgrand, 2025-01-28:
> Same issues here. Firts i think i burned the rf module, changed it (85€), then buy twin little pro and twr8, and power is about 75db at 2meters, either internal and external, 900 band is fine, lora [...]

I did a quick check to 15m in my garden in less than ideal conditions, Rssi 84dB. 
Transmitter is on 1.5.9 Flex.     Receiver, TW GR6 is on 1.0.6.     Twin lite pro module set to 100mW rf output power.
I am waiting for 1.6.1 to be released before I run the updates again.

drewsho, 2025-01-28:
Hello,

Now I would like to put forward a few wild theories on the subject of RSSI after the update to Ethos 1.6.0 in connection with the update of the RF components to version 3.0.1.

I have now checked my Twin Lite Pro module and found that it transmits with both protocols.

So I would like to put forward the following theory:

In Ethos version 1.6.0, both RSSI values (LoRa and FSK) were not yet displayed in the status bar. However, with RF firmware 3.0.1, both RSSI values are already transmitted separately.

Because with the old firmware 1.0.7 only one RSSI value was transmitted, and here presumably the value of LoRa, or a combined value, this value was of course always displayed in the status bar.

With the new RF firmware 3.0.1, a new RSSI value was added. However, the status bar showed the value that had always been transmitted.
My guess is that the added sensor contains the value for LoRa and the existing sensor contains the value for FSK.
That's why version 1.6.0 only shows the values for the FSK value and this is significantly lower and therefore messages about RSSI problems come up relatively quickly with Ethos version 1.6.0.

Until then, all just wild guesses, but they would sound logical.

I then installed Ethos version 1.6.1 RC3 (I actually didn't want to do that before the official version was available).

Both RSSI values are now displayed in the status line and even if the RSSI value goes from FSK to 0, there is still no warning because the LoRa value is still OK. This behavior can now also be adjusted in the telemetry settings (individual RSSI alert per band).

This is the only way I can imagine this confusion with the RSSI values.

I also did a range test yesterday at the airfield with full power (i.e. not in "range test" mode). With visual contact to the model, I still had values of 51db FSK and 84db LoRa at a distance of around 550 m. Only when I lost visual contact through the forest and a mountain ridge a few hundred meters further on did the values go to zero and the receiver went into failsafe.

My conclusion is that the problems only occur in connection with Ethos version 1.6.0 and RF firmware 3.0.1. With version 1.6.1 RC3, the FSK values still decrease quickly, but this no longer leads to RSSI warnings.
I think that was the case with previous versions too, and since it was not displayed, nobody was interested.
I don't want to encourage anyone to be careless and so I would recommend that everyone do a range test before their first flight.

jereitor, 2025-01-29:
I have the same thoughts as you, I have also put the RC3 version and everything seems to work better, but the RSSI of FSK seems to me very low compared to lora, although I dare to fly because the Lora signal seems to be consistent.

jereitor, 2025-01-29:
> I have the same thoughts as you, I have also put the RC3 version and everything seems to work better, but the RSSI of FSK seems to me very low compared to lora, although I dare to fly because the Lora signal seems to be consistent.

Sorry, I accidentally closed it

clintc2000, 2025-02-01:
[image]

[image]

[BAMF2-Green-2025-02-02-12-08-52.csv](https://github.com/user-attachments/files/18630107/BAMF2-Green-2025-02-02-12-08-52.csv)

got out for a few minutes today.  never got more than 150m away from plane.  Brand new TWR6, same results as the TWGR6.  Log file show both ground range test and a couple minutes of flying.  ground test was about 70m.  Flying about 150m and 30m altitude at max.

kramgrand, 2025-02-02:
That sounds like the results I had before I replaced the TW Lite Pro module.    Critical rssi issues at about 150m range.     It had worked fine until I updated the module to the latest firmware.

I have since updated the replacement module to 3.0.2 and all is good.    Try and get hold of an ImmersionRC RF Power Meter. That is how I proved my module was faulty.

IMG_5896.jpg (view on web)<https://github.com/user-attachments/assets/f27c91da-4d2f-48c3-a72c-63af153b4563>

IMG_5898.jpg (view on web)<https://github.com/user-attachments/assets/0661d10c-2ea1-4146-965d-6236b5796a9e>

clintc2000, 2025-02-02:
I do see a difference when I increase the tx module power output, and if I shield the FSK antenna.  So I would think that would indicate power is making it out the antenna.   Otherwise it shouldn't make a difference if I change power output or shield (wrap my hand around just the antenna).  Or am I thinking wrong?

jereitor, 2025-02-02:
you quote version 3.0.2 , where did you get it from, or do you mean RC2 ?

clintc2000, 2025-02-02:
downgraded just the module to 1.0.7 and everything still works.  was able to fly several hundred meters without issue, even at 10mw.  Don't know if it's masking a bad tx on fsk or if issue is just with 3.0.1

Anyone have any ideas on how to tell, besides removing a tx antenna (don't want to risk damage) or an expensive power meter?

Burgerman-X10express, 2025-02-03:
A cheap ebay microwave "leak" detector.

kramgrand, 2025-02-03:
> 
> you quote version 3.0.2 , where did you get it from, or do you mean RC2 ?

The 16th post from the beginning from SmartTommy  two weeks ago has the down load link for 3.0.2 TW module firmware.

kramgrand, 2025-02-03:
I know the rf meter is quite expensive, but to my mind it has paid for its self since I have had it and it proved without a doubt that my TW module was faulty. That enabled me to get a warranty replacement.  If it was the update I can't prove either way. Just seemed a bit coincidental.

Burgerman-X10express, 2025-02-06:
I am going to try and give you an explanation. It may be wrong!
I will use the 900 and 2.4 TD recievers as a clearer way to explain.
The same aplies to the twin 2.4 systems but is much less pronounced.
Remember range is determined by a few things. One is reciever sensitivity. And an a twin system thats much lower than on the LoRa half of the system.

On these systems the amount of latency, and the amount of accuracy in the positional response affects the range too. Its how they get better sensitivity. Its how the lora system peices together useful info from noise at a distance. 
Because to be super accurate with many channels and to have low latency too requires a very good strong signal.

The LoRa longer range systems have less accuracy, and are a little slower as they are working at Longer Range (lora) so need to collect more packets of data to get an adequate signal from the noise.
Since most of these recievers are using a single antenna per half of the system, that only works in one plane (with a null at either end) then then the shorter range faster system gives you less distance depending on the angle of the antenna as well as less sensitivity. This likely happened in the past but you never knew as the other antenna was at 90 degrees to it. So one or the other always saw a signal. But now theres only one antenna per "system". 

Because the lora side, has better range anyway, especially on the 900 systems, its always easily possible to fly well out of range of the 2.4 system, by simply miles. Because that 900 lora typically got 20 or 40 miles of range. So even the null points at the ends of the antenna are no problem at a few miles.

jamesn1960, 2025-02-07:
> CAUTION: This email originated from outside the organization.

Thank you for your response and explanation.

Note: I am only referring to the degradation of RC transmission range experienced after converting from the FrSky X-Lite Pro (OpenTX) and G-RX6, G-RX8, Archer GR6, Archer GR8, and the Archer Plus GR6&8 receivers 'vs' the so called superior FrSky Twin X-Lite S Transmitter (Ethos) and FrSky TW GR6 and TW GR8 receivers.

I also use TBS Crossfire in a lot of my RC INAV planes and Quads since TBS Crossfire was introduced to market.  With my 24 year employment one of our leading global products also use LoRa 900 for long range communication.  So I do have a general understanding of the LoRa technology.

My question was, FrSky advertise the Twin 2.4 system as being a better system vs a single band 2.4, and that a dual 2.4 system offers longer range.  However, to date in my DLG gliders I have found the Twin 2.4 system to be far worse then the original FrSky single band 2.4 system.  In my experience switching to the FrSky Twin system has been a substantial set backwards, my overall experience so far has been very disappointing and a waste of my hard earned cash.

However, what I read into your response is that a can comfortably ignore the RSSI Critical warnings my Twin X-Lite S transmitter is constantly screaming out when less then half the distance of what I once obtained from my X-Lite Pro transmitter and single band 2.4 receivers?

As mentioned, apart from switching to the FrSky Twin system nothing else has changed, I fly at same location so there is no change re possible interference, and the new Twin systems are used and mounted/installed identical to all my previous arrangements.  God help me what I will experience if I attempt to use my X20S Tandem with the Twin module I also purchased for it.

Burgerman-X10express, 2025-02-07:
> However, what I read into your response is that a can comfortably ignore the RSSI Critical warnings my Twin X-Lite S transmitter is constantly screaming out when less then half the distance of what [...]

Pretty much.
Why not use the module with the 500mW output as that will increase signal strength at every place in space by double, and also double the range. The lora part should give you 20km or at least it would if not using a carbon fusalage.

Question.
Do you use the antennas with the small brass tube or just the bared cable end?  Those are a lot worse. 
Do you make sure some of the coax is external as with the simple stripped cable type prefered by carbon glider pilots the coax is a part of the antenna.
Do you make sure that at least one antenna points down and your TX transmitter points up? So that it is seen visibly by much of the flight and is in the same plane as the TX antenna?
Do you make sure that the end of the antenna IS NOT taped to the side of the conducting (like a metal) fusalage as that will severely mess up the antenna performance and put its SWR sky high, and out of tune?

jamesn1960, 2025-02-07:
Thank you for quick response,

Do you actually know your own FrSky products? it is NOT possible to increase the signal strength of the 'Twin X-Lite S transmitter'  If it is possible please explain how, because there is no field within ETHOS to increase the modules power in a Twin X-Lite S transmitter.

I am aware that you can increase the power of the TWIN Pro TX module, but I am not referring to this product.  I have not yet tested my Twin Pro module with my X20S Tandem transmitter and have no intension to do so until I feel comfortable with the FrSky TWIN system.

There is no way possible that I am able to obtain 20km range you mention when using my new Twin X-Lite S transmitter and TW GR receivers.  I'm lucky to obtain 100 meters with this new TWIN system.  Prior to changing to the Twin system I could easily obtain 360 meters with my X-Lite Pro transmitter and the standard GR and Archer GR receivers, beyond this range my poor old eyes were lucky to see my 1.5mtr DLG gliders.

Below are three images (taken just now) of how my TW GR6 and TWGR8 receiver antennas are mounted in my DLG gliders.  NOTE: how they are mounted in the images is exactly "IDENTICAL" to how the antennas were mounted when using FrSky standard GR and Archer GR receivers.

With the TWIN GR6 & GR8 receivers my max range is 100 meters before receiving RSSI critical warnings. Prior to the TWIN system I couldn't see my DLG gliders before I would receive RSSI warnings?

[IMG_1505.jpg]

[IMG_1506.jpg]

[IMG_1507.jpg]

clintc2000, 2025-02-07:
I think you indicated you were using ethos 1.6 with rx fw of 1.0.6?  Can you verify fw of the tx and rx?  Should be 3.0.1 on ethos 1.6That said, pretty sure there is a bug with twin, ethos 1.6 and fw 3.0.1I’ve not had lose of control, but you do get warnings too close.

jamesn1960, 2025-02-07:
Yes correct re ETHOS firmware, it is 1.6 and receivers 1.0.6.

Although I have owned 2 x Twin X-Lite S transmitters for some time now I have only just started using them for my DLG’s.  I had been waiting for Mike Snabaitis – RC Soar to design a Ethos template for 6 servo DLG’s.  Before using the system in the field I updated my Twin systems to the latest firmware.  For my F5J and scale gliders I’m using non TWIN receivers and my X20S Tandem transmitter.  The RSSI Critical issue only exist with my TWIN systems in DLG gliders.

If it is a firmware issue hope you guys can sort it out soon.

Cheers, James

clintc2000, 2025-02-07:
Yeah James, I'm same as you.  Been using Twin Xlite S for DLG for 2 years, both with and without twin pro module.  you could try ISRM 1.0.7 on the tx, and 1.0.6 on the rx.  I suspect if you updated to Ethos 1.6.0, your tx ISRM is now 3.0.1, and rx is still 1.0.6.  I've not tried that combo.  Though I did bind my rx @ 3.0.1 to my TX on Ethos 1.5.11(?) and ISRM 1.0.7 and it seemed fine, though I didn't fly with that combo.

clintc2000, 2025-02-07:
More testing results.  I have a good friend with a immersionRC RF meter.  I was able to verify my twin pro modules are good.  Output was similar between both antennas and both modules on both 3.0.1 and 3.0.2 fw for the module.  And also similar to my module on 1.0.6 fw (Ethos 1.5.18)

Definitely points to this being a software issue, maybe on the rx fw.

[image]

[image]

[image]

[image]

jamesn1960, 2025-02-07:
Thanks Clint, I just connected my Twin X-Lite S to the Ethos suite, there's a new update today so I will be updating now and hopefully get to test this weekend, that's if the rain holds off.  Here in northern QLD (tropics) Australia we're lucky to have had 1 week without rain since mid-Dec, past couple of weeks it's been torrential, flooding, monsoonal cyclonic weather.

My Twin X-lite S is presently on 1.6.0 FCC, bootloader 1.4.16, Audio 1.5.18, RF module 3.0.1.  My Twin receivers are on 1.0.6.  These are the firmwares I have been having issues with re RSSI Critical warnings from 100 meters.  Prior to this I can't not comment because I had not previously used the FrSky TWIN system.

Ethos suit today saying firmware and audio out of date, so I just now hit the update tab.

James

clintc2000, 2025-02-07:
I'm updating to 1.6.0 as well.  You should update the rx to 3.0.1, but I don't think it will help with the RSSI alerts.  

Weather here in Utah has been wet or really cold (sub 0 cel) on weekends.  Only been able to get out a couple of times for limited flying.  Contest in 2 weeks in sunny Florida.  Hoping the fw issue gets resolved before then, or I may just go back to 1.5.18 and 1.0.6

jamesn1960, 2025-02-07:
Hi all, at the same time I also contacted FrSky tech department asking the same RSSI Critcal warnings.  Below is the response they just now provided.

Dear sir

Please wait for receiver and RF module firmware release to v3.0.2.
This problem have been reported with v3.0.0 and v3.0.1.

A temporary solution:

1. Radio RF module downgrade to v1.0.7 for twin lite pro.
2. Receiver firmware downgrade to v1.0.6 for TW GR6.
3. Receiver firmware downgrade to v1.0.9 for TW GR8.

IMG_5906.jpg (view on web)<https://github.com/user-attachments/assets/a5fd5d0e-d05d-4551-b3f5-9d3da68575b0>

IMG_5907.jpg (view on web)<https://github.com/user-attachments/assets/47c002c8-9983-4456-be02-e9abc9d96fcf>

IMG_5908.jpg (view on web)<https://github.com/user-attachments/assets/452cd699-5c71-48f5-b362-76207150a017>

IMG_5911.jpg (view on web)<https://github.com/user-attachments/assets/a0765e83-f5cf-428d-9c90-785982fc3cc2>

-

Burgerman-X10express, 2025-02-07:
> Do you actually know your own FrSky products? it is NOT possible to increase the signal strength of the 'Twin X-Lite S transmitter'  If it is possible please explain how, because there is no field within ETHOS to increase the modules power in a Twin X-Lite S transmitter.

I was refering to my OWN system, which can do both tandem and dual internally. And I have an additional MODULE that works the same as your transmitter added as well.  Albeit the power output may be more limited. So my transmitter (which has the X20 pro) has 2 seperate ways to do the exact same twin receivers.

On that I can choose many different power levels. But even if you cant do that with your TX you can with your module. All the princibles and firmware numbers and recievers remain the same.

kramgrand, 2025-02-07:
A temporary solution:

1. Radio RF module downgrade to v1.0.7 for twin lite pro.
2. Receiver firmware downgrade to v1.0.6 for TW GR6.
3. Receiver firmware downgrade to v1.0.9 for TW GR8.

**Does anything else require rolling back?**  Using XE-S on Flex 1.6.1   TD-ISRM 3.0.1

jamesn1960, 2025-02-07:
Thank you Burgerman, at this stage I have not yet attempted to use my Twin Lite Pro module on my X20S, and I have no intensions to until the fault has been resolved, I will continue to use the Access protocol on my X20S so that I'm able to continue flying models.

For now I am ONLY referring to my personal experience with the FrSky Twin X-Lite S transmitter and the TW GR6 & 8 RX's.  Like others I have also tested the RC output with my ImmersionRC RF meter and the output on the latest firmware is not as a good as the Access protocol.

Having so many RC systems and models I am now dealing with TBS Crossfire because their Nano RX's haven't the memmory capacity to hold their latest firmware update.  So for the FrSky Twin protocol I intend to hold back until FrSky can resolve their problem vs going through the whole process of rolling back firmware on all my Twin systems.

jamesn1960, 2025-02-07:
Hi Kramgrand,  May I suggest contacting ***@***.******@***.***> regarding your question, he only provided the temporary solution as noted.

Burgerman-X10express, 2025-02-08:
To add some information.
I now have 2 brand new GR8 recievers.
Both of these display 100 RSSI on FSK either with the module or the TW transmitter.
Both refuse to display above 75 to 76 RSSI even at 12 inches range.  With transmitter, or external module.

So theres something wrong with the recievers firmware.  Radio and module are all of latest official firmware.

Burgerman-X10express, 2025-02-08:
>>>For now I am ONLY referring to my personal experience with the FrSky Twin X-Lite S transmitter and the TW GR6 & 8 RX's. Like others I have also tested the RC output with my ImmersionRC RF meter and the output on the latest firmware is not as a good as the Access protocol.

Warning...
Measuring feild strength doesent tell you the output level as it averages the data much like a voltmeter. An example.
On spectrum DSM or DSM2 shows a much much greater field strength using a Feild strength meter. Compared to DSMX. Why? Because DSM and DSM2 transmit 2 channels at 100% of the time. DSMX transmits on different channels only very short under 2% of the total time in pulses. Both have the exact same peak output and range. With a 50x difference in measured feild strength. So the protocol is at least as important as the mW.

Burgerman-X10express, 2025-02-08:
Fitted 2 new lora antennas, both receivers. Perfect. TWO faulty out of two. I think the antennas were OK just those stupid tiny connectors were not making contact. I dont believe it myself. But now have 100rssi on the lora as expected. All normal latest firmware..

kramgrand, 2025-02-10:
Hello everyone.  This the latest update I have received from FrSky Tech Support.

On 10 Feb 2025, at 16:30, Service Engel Modellbau <service@engelmt.de> wrote:

Hello Mark,
 
I can't give you any definitive information or advice at the moment. Except possibly flashing ETHOS back to 1.5.19 and RF 1.0.7.
I am in contact with FrSky about this. There will probably be an RF firmware 3.0.2 which should solve this problem. But I can't say when yet.
Have you ever done a range test? That is, until the warning appears and then until Failsafe appears? From this you could conclude whether it is a real range problem or just a problem with incorrectly displayed RSSI values.
But to make a long story short. FrSky knows that the problem exists and is working on it.
Mit freundlichen Grüssen
With best regards

Engel Modellbau & Technik

jereitor, 2025-02-10:
> Hello everyone. This the latest update I have received from FrSky Tech Support.
> 
> On 10 Feb 2025, at 16:30, Service Engel Modellbau [service@engelmt.de](mailto:service@engelmt.de) wrote:
> 
> Hello Mark,
> 
> I can't give you any definitive information or advice at the moment. Except possibly flashing ETHOS back to 1.5.19 and RF 1.0.7. I am in contact with FrSky about this. There will probably be an RF firmware 3.0.2 which should solve this problem. But I can't say when yet. Have you ever done a range test? That is, until the warning appears and then until Failsafe appears? From this you could conclude whether it is a real range problem or just a problem with incorrectly displayed RSSI values. But to make a long story short. FrSky knows that the problem exists and is working on it. Mit freundlichen Grüssen With best regards
> 
> Engel Modellbau & Technik

At last someone is saying that FRSKY is aware of the problem. Hopefully this will be resolved quickly, I have some of my models on hold waiting for version 3.0.2.

Joshadam0057, 2025-03-06:
Anything being done about this?

jereitor, 2025-03-06:
> ¿Se está haciendo algo al respecto?

nothing seems to be moving

[7 more replies]