FrSkyRC/RF_Feedback_Group #107: Not seeing Rx VFR on TD R18

lthole, 2025-01-27:
Ethos 1.6.1 RC3 on X20
TD ISRM 3.0.1
TD R18 3.0.1 note .frsk file dated 2024-12-11 downloaded today 2024-01-27 from Frsky TD R18 download page
Performed factory reset on TD R18, then registered and bound.

From System/Info:

[image]

[image]

From Model/RF:

[image]

From Model/Telemetry with Individual RSSI alert OFF: 
Note no Rx VFR.

[image]
Note I sorted the sensors below:

[image]

From Model/Telemetry with Individual RSSI alert ON:
Note still no Rx VFR. And for some reason 2 instances of RX 0 sensor.

[image]

[image]

SmartTommy, 2025-02-05:
Hi, it my fault, i forget send the RX VFR to the Ethos. fixed.

[TD-ISRM_3.0.2_RC1.zip](https://github.com/user-attachments/files/18665327/TD-ISRM_3.0.2_RC1.zip)

Alex00101, 2025-02-05:
Thanks is there a 3.0.2 update (rc) for other modules ie TD-ISRM Pro?

SmartTommy, 2025-02-05:
> Thanks is there a 3.0.2 update (rc) for other modules ie TD-ISRM Pro?

The TD-ISRMPro and Rs send the RX VRF normal. 
the PRO AND RS 3.0.2RC1 fixed the issue : 2.4G RSSI no change when RX power off  

[TD-ISRMPro And Rs.zip](https://github.com/user-attachments/files/18665992/TD-ISRMPro.And.Rs.zip)

SmartTommy, 2025-03-06:
3.0.2 fixed

closed 2025-03-06 by SmartTommy